keirf/flashfloppy #242: FF v3.1a - FF-CFG setting oled-font and oled-text question

bjkock, 2019-06-21:
Hi!
I´m trying out the experimental version 3.1a
I have an OLE 128*64 - 0.96” and really like the new 3 row function on the display, but It behaves strange if I in the FF.CFG use the combination:
oled-font = 8x16
oled-text = default
When I use this combination the first row is displayed as double height 8x16, as I suppose it should, but row 2 and 3 gets cut and only displays half the text (top half) in single height.

[image]

When I use the combination:
oled-font = 6x13
oled-text = default
The first row is displayed as double height 6x13 and row 2 and 3 is displayed as single height and all the rows is fully readable.

[image]

Is this a known and intended limitation or is it a bug (or maybe me doing something wrong)??
// Jesper Kock

keirf, 2019-06-21:
You should have `display-type=oled-128x64` in FF.CFG too?

bjkock, 2019-06-21:
Yes, I have display-type = oled-128x64-rotate in the FF.CFG.

This is the settings in my FF.CFG:
interface = jc
host = unspecified
pin02 = auto
pin34 = auto
write-protect = no
side-select-glitch-filter = 0
track-change = instant
index-suppression = yes
head-settle-ms = 12
motor-delay = ignore
chgrst = step
ejected-on-startup = no
image-on-startup = last
display-probe-ms = 3000
autoselect-file-secs = 5
autoselect-folder-secs = 10
folder-sort = never
sort-priority = folders
nav-mode = default
nav-loop = yes
twobutton-action = zero
rotary = full
indexed-prefix = "DSKA"
display-type = oled-128x64-rotate
oled-font = 6x13
oled-contrast = 200
oled-text = default
display-off-secs = 200
display-on-activity = yes
display-scroll-rate = 300
display-scroll-pause = 1000
nav-scroll-rate = 160
nav-scroll-pause = 600
step-volume = 20
da-report-version = ""
extend-image = yes

keirf, 2019-06-21:
What happens if you get rid of the -rotate suffix?

keirf, 2019-06-21:
And has this display worked properly in v2.x? The scan line appearance in both your pictures is wrong. It shouldn't look like that, even the more legible 6x13 one.

bjkock, 2019-06-22:
Well....Maybe there is a problem with the OLED....
I reset to factory default configuration, "downgraded" back to FF v2.13 and used a FF.CFG that only hade the row: display-type=oled-128x64 and put it in the root of the USB-Drive.
After it read the config file there is no change in the appearance of the screen at all.
I changed the FF.CFG to only contain the row: display-type=oled-128x32 just to see if the screen appearance changed, but it still looks exactly the same.
To be certain that the FF.CFG file got read I then changed the row to: display-type=oled-128x64-rotate, this makes the screen turn and I guess that means that the file is read correctly.

This is how the screen looks with either of the commands display-type= oled-128x32 or display-type=oled-128x64.

6x13
[image]

and with oled-font = 8x16
8x16
[image]

Any ideas what I can try next?
Or should I just get another screen and see if that solves it?

keirf, 2019-06-22:
Maybe the display is unsupported. There are various sorts available. Can you attach a photo of the pcb and a link to the sale listing?

bjkock, 2019-06-22:
Here are photos of the front and back of the display.
[image]
[image]

and the link to the E-Bay page witch i purchased it from:
https://www.ebay.co.uk/itm/0-96-Zoll-Arduino-OLED-Display-SSD1306-BLAU-128x64-I2C-IIC-Modul-SPI/253048439903?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

keirf, 2019-06-22:
I have one just the same, it's definitely supported (ignore the refresh artefact). I guess yours is bad, I see the corner of the glass is smashed. Not good for a circuit-on-glass. ;) If it came like that get a refund.

[image]

closed 2019-06-22 by keirf

bjkock, 2019-06-22:
Oooohhh, is that the way the text should look.
Well, I guess I just have to buy another one and hope I have better luck this time :-)
I actually didn't notice the missing piece from the corner of the glass until you mentioned it, so I don't know if it came that way or if it is me that cracked it :-o

Thanks for your time and help, much appreciated!!!

bjkock, 2019-06-24:
Hi Again!
Found a thread about these 128x64 displays and "row skipping" issues like mine have.
This is definitely a hardware failure that happens when the lower left or right corner of the glass is broken :-O
So I have placed an order for a couple of new displays and is confident that this will solve the problem.
:-)