keirf/flashfloppy #243: MS-DOS 6.22 Startup Disk

koklica, 2019-06-23:
Dear Keirf
I am using 26 pin slim floppy, everything is normal USB main directory and can read and write in the ADF and CFG normal. My system is DOS 6.22 system and I can not create a startup disk (USB) can not start with this floppy disk (USB) I can not help me on this.

keirf, 2019-06-23:
Just copy a 1.44mb IMG file onto the stick? Jumpers at S1 and JC. You probably won't need a FF.CFG.

koklica, 2019-06-23:
So 6.22 MS-DOS. IMG will not be ff. CFG to the main directory

keirf, 2019-06-23:
Forget about FF.CFG. For now just put one IMG file on the USB stick. No other files. It should be automatically recognised and mounted by Flashfloppy. Then you should be able to access it from your host PC. Even boot it if its a bootable disk image.

koklica, 2019-06-23:
OK I'll try tomorrow and report the result

koklica, 2019-06-24:
Thank KEIRF
I tried the main directory succeeded DOS. img
ff. CFG does not exist

closed 2019-06-24 by keirf