keirf/flashfloppy #244: Gem Genesys Pro and "Disk not insert”

andr229k, 2019-06-25:
I struggle with this Gem Genesys Pro and it seems stubborn.

**Installation:**
Jumper on S1
Interface setting = ibmpc [[Link](https://drive.google.com/open?id=10sCkUF6uSu1kAl6QbefduZCn0nh7cFj-)]

**Background:**
The owner used a Gotek with the Chinese factory Firmware (No FlashFloppy or HxC). It uses a jumper on S1 and it works fine. The user is just sad that it don’t have OLED and rotary compatibility like HxC and FlashFloppy.

I asked him how he operated the factory firmware with his Gem Genesys:
```
1. He unpack a new USB key 
2. Insert the usb key in the Gotek. (without any USB-stick preparation) and scroll to the Gotek-LED number he wants to save on. 
3. Gem Genesys Pro generates a folder with the desired content called (example D0173B7-144, D0175FA-144, D018BA0-144) (Those names are the name-convention for the different LED-number..banks..000..001..002..etc) (I guess the 144 in the name stands for 1.44 mb)
4. User can then unplug his USB-stick from the Gem Genesys and plug it into his computer and actually see the raw content, which is saved on the stick.
```

**Example of disk image the factory firmware makes** [D018BA0-144]: 
[[Link](https://drive.google.com/open?id=1IVL8f2jC1gymIyOAh3ljKY02xQENPFMf)]

**Note:**
- I tried generating a 1,44 MB image with HxC Floppy Emulator software, but the Gem Genesys display gives the end-user "Disk not insert”, when user tried to access the image.
- I was told that support for "soft" images for FAT folders could be a solution.

keirf, 2019-06-25:
Jumpers at JC and S1. Don't bother with an FF.CFG for now. Reset flash configuration by holding both buttons for a few seconds with no USB stick inserted.

A blank 1.44M IMG ought to be recognised as inserted. Perhaps it did not like the IMG if it was not FAT12 formatted?

No support for FAT folders at the moment. I will work on that.

andr229k, 2019-06-25:
Hi,
Keir thanks for the prompt reply. I had to make the topic here due to that the end-user didn’t have Facebook.

We have only tested a blank .HFE 1.44MB DSHD, not IMG.

=====
Advise end-user to set a jumper on JC and S1.
Reset flash configuration
Provide end-user a 1.44MB DOS .IMG [Link]]([Link](https://drive.google.com/open?id=12Ea_wFTJDAnsjDnV2BU9QGcumqU7Bxaz))
=====

Again thanks, I will get back with the feedback.

keirf, 2019-06-25:
HFE should work no problem, but anyway, I trust an exactly 1.44MB sized IMG more :)

keirf, 2019-07-07:
Is there any more to be done on this ticket?

andr229k, 2019-07-07:
> Is there any more to be done on this ticket?

Hi Keirf,
we are still in some kind of test phrase. 

We were able to successfully format a 1.44MB DOS image on the Gem Genesys. There is on the other hand something strange. User duplicated the image and some fails and some successfully format in an unpredictable way. (so, something is not 100 % correct)

Are there any development on following?: 
1.	on "soft" images for FAT folders? (This is how the factory firmware works and factory firmware seems to work better with these kind of devices because the items are not packed into some kind of nut/shield image)
2.	Are there any plans for backward compatibility from the factory firmware? (So, you could read content on FlashFloppy made from the original factory firmware)

keirf, 2019-07-07:
If you have one empty formatted image, best approach may be to clone/duplicate that image every time you need a new freshly formatted disk. Latest v3.x firmware has z clone feature built in, or of course you can easily plug the stick into a PC and make copies of the formatted image there.

No work has been done on soft folders yet. But bear in mind that the factory firmware turns the folder into a FAT12 IMG file when the folder is mounted, and so it's not so different from Flashfloppy after that point anyway.

There is no specific intention at all to support factory-firmware USB stick layouts.

keirf, 2019-10-01:
Closing due to lack of activity.

closed 2019-10-01 by keirf