keirf/flashfloppy #248: Hardware question - rotary encoder mod

roses2uk, 2019-07-07:
I refer to the two diagrams on your Wiki. 

Part 1) My rotary encoder arrived recently. It has a backboard on it with a few 10k resistors and five connector pins: CLK, DT, SW, + and GND. Apparently you have to connect the + to 3.3 V (pin 2, adjacent to J3 header in front of the power connector. But with the backboard, it is clearly too physically big to fit into the Gotek case, so it means removing the backboard as per drdpj's YT video 'FlashFloppy Gotek mods (11m30s onwards). But doing that means there are now only four connections: CLK, DT, SW and two GND's which he connects together and attaches to the front of the JA connector of the main Gotek jumpers. So what I don't understand is where does the rotary encoder get its power from now as there is no + connection?? 

Part 2) drdpj connects and solders both grounds together. For simplicity, is it possible for me to have two separate cables, one to the front JA pin, and the other...to where? Are all the front connectors of the jumpers strip grounds? 

Thank you.

drdpj, 2019-07-07:
I already answered part 1 when you asked in the comments section on YouTube. Link the grounds, it's by far and away the easiest way.

roses2uk, 2019-07-07:
Yes, you said "The grounding is what's detected as the pin on the uC will end up sitting at 0V instead of 3.3V. You don't have to worry about it really, just know that if you've got the resistors, you need the 3.3V connection, and if you don't you just connect it directly." But I didn't really understand (don't have an electronics degree or years of experience doing this kind of work) so was trying to seek alternative clarification from Keir. But OK, I'll take you at your word and just follow the instruction blindly - if I connect directly after removing the backboard, no connection to 3.3V power is necessary. I appreciate your input.

keirf, 2019-07-07:
See the second diagram in the Rotary 
Encoder section of the wiki: https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods#rotary-encoder

This is a pictorial representation of exactly what drdpj has already said, in video and comments.

ie. he is correct: link the grounds, and there is no need for 3.3v power when the pcb is removed.

closed 2019-07-07 by keirf

roses2uk, 2019-07-30:
Re: the second diagram, just to clarify - the view is taken from the REAR of the encoder with the shaft pointing away from oneself, correct?? So clockwise from top right: CLK, GND, DT, GND (bottom left), SW (top left). A replacement encoder arrived today, although confusingly the pins on backboard say GND, S1, S2, Key, 5V. But when I take the backboard off, there are still 5 pins as per your diagram. Thx.

keirf, 2019-07-30:
The diagram is actually from overhead but actually the pin order doesn't matter as much as it appears.

On the side with two pins connect one to JA and the other to GND. It doesn't matter which way round.

On the side with three pins, middle is GND. The other two can be connected either way round and simply swap them if the rotary is working backwards.

roses2uk, 2019-08-02:
As a test, I connected my encoder (with its backboard) to the Gotek: GND to front JA, S1 to back JA, S2 to back JK, KEY to front JK. Is that right? It seems to work, except both clockwise and anticlockwise rotation increments by one - there is no backward step possible, and I couldn't find a config file command to correct this. My mate will come by this weekend to desolder the backboard and solder the 5 wires to the encoder's tags. Thx again Keir and drdpj.

keirf, 2019-08-02:
The backboard may affect operation. Best wait and see with it removed

roses2uk, 2019-08-05:
Yep, you were right, without the backboard it all appears to be working correctly, so I'm a VERY happy camper. One thing though: is there a config file command so that pressing the encoder switch goes up one level (..) in the folder tree?

keirf, 2019-08-05:
No but you can do it with the original two buttons with suitable setting of twobutton-action in FF.CFG. Look in the example FF.CFG for explanation

roses2uk, 2019-08-07:
Got it!! Added the line 'twobutton-action = rotary' to FF.CFG. Now my plan is to customise the OLED display to show the information that  *I want*. For me, the 'T:00.0' track countdown is unnecessary and superfluous. How do I remove it?

keirf, 2019-08-07:
The printf statement is in src/main.c. You can remove or replace it

roses2uk, 2019-08-07:
Aah, there's no config file command?? OK, doing "printf in src/main.c" doesn't mean anything to me and is therefore beyond my technical limitations. Guess I'll have to live with the display contents as it is. Thanks anyway...

keirf, 2019-08-07:
There are some display options in the latest v3.x configuration. So you can choose what is displayed on each row, from a small menu of possibilities. Take a look.

roses2uk, 2019-08-08:
Still not sure if I can get my head round it. I'll wait until a stable 3.x is issued...