keirf/flashfloppy #249: Only 310 slots

franeti, 2019-07-14:
Hi!

I have the gotek with ff v2.13. I only have 310 slots available to assign. Is not possible to get more? I thought I have 999. The thing is that I would like to assign more than 310 disks.

Do I have a problem in my unit or is it possible to configure this?

keirf, 2019-07-14:
I assume you are using FlashFloppy with an OLED/LCD display, and not using the AUTOBOOT.HFE file selector?

By default FlashFloppy sorts folders into alphabetical order. If there is insufficient space to do this in full then it will only list part of the folder. If you are only achieving 310 slots then your image names must be quite long (TOSEC?)!

Anyway, a couple of options, if my assumptions about your setup are correct:
1. Place images into a folder hierarchy. For example, A-I, J-Q, R-Z. This kind of arrangement obviously reduces the number of images per folder.
2. Disable folder sorting in FF.CFG. See the folder-sort option in https://github.com/keirf/FlashFloppy/wiki/FF.CFG-Configuration-File

Let me know whether this helps, and whether my assumptions are correct.

franeti, 2019-07-14:
Thanks for your quick reply.
Yes, I´m ussing the autoboot.hfe file selector. HxC Floppy Emulator file selector v1.89a.

All my files are in a folder "games" in the usb stick. Anyway, the issue when I´m assign it. There are "only" 310 slots. I have more than 310 files to assign.

keirf, 2019-07-14:
Please use the AUTOBOOT.HFE and HXCSDFE.CFG files from the FlashFloppy distribution. They are in the HxC_Compat folder. Place the HxC_Compat/Amiga/AUTOBOOT.HFE and HxC_Compat/HXCSDFE.CFG files in the root of your USB stick, overwriting your existing versions. Then see how you get on.

franeti, 2019-07-14:
Ok, now it works perfectly. Thanks for your support.

Regards.

closed 2019-07-14 by keirf