keirf/flashfloppy #250: not working on some newer goteks from china

sub205, 2019-07-15:
Hello,

i got some goteks from 2 sellers from china. Flashfloppy installed fine, but the connected computers cannot read / identify the disks. The Goteks work fine with their stock firmware and the flashfloppy-image itself works fine on older goteks i have. It seems to be something at hardware level, although both PCBs look identical to me. I already tried several jumpersettings. One drive worked one time in a ibm portable p70 for a short time.

Same problem for older firmwares, tried several.

I cannot see a reason for this, has anyone else experienced this behaviour?

Will try to change the µCs to rule out a fake controller but cannot imagine that one would fake these chips and they still work with original firmware. Right now, 3 drives are unusable because the flashfloppy fw isn't working and i cannot go back to gotek fw.

omikron88, 2019-07-15:
May you make the detailed photos of both sides of PCB, please?

sub205, 2019-07-15:
Will upload pictures later that day! Currently out of lab.

keirf, 2019-07-15:
More likely they are all duffs or you have connected them all in such a way (eg. new system?) that they got systematically fried. 

There is an I/O Test firmware you can update to (via USB stick, no need to re-flash from scratch) which allows you to test all inputs and outputs at the floppy header with a single jumper wire. See here: https://github.com/keirf/FlashFloppy/wiki/Testing-IO-Pins

sub205, 2019-07-15:
I tested several systems, controllers and drives, swapped forth and back, tested a fresh gotek of the new batch with stock and with flashfloppy firmware to reproduce the problem. There is 100% nothing fried! I tested 286, 386, pentium3, ps/2, at least 6 different systems and controllers.
I can narrow it down to the newest batch of drives i got. I also flashed from scratch (like i did with my first, working drive) and also upgraded all drives (including the working one) the same way, just so have the error persistent. That's why i'm mentioning it here.

I'll test the IO firmware tomorrow and report.

keirf, 2019-07-15:
Okay, I/O Test and high-def photos of the PCB are both good things anyway. As well as testing the functionality of I/O pins, it can also indicate if assignments to MCU I/Os have changed.

keirf, 2019-10-01:
Closing due to lack of activity and no other user reports.

closed 2019-10-01 by keirf