keirf/flashfloppy #251: no issue - but featurerequest: OSD

fook42, 2019-07-15:
i read, that on HxC an OSD (on screen display) feature is available.
maybe this can be implemented into FlashFloppy as well? of course with a Hardware-Mod (connection to VGA/composite)

fook42, 2019-07-15:
additional information may be find here:
http://torlus.com/floppy/forum/viewtopic.php?f=33&t=1683&p=20587#p20587

fook42, 2019-07-15:
it works like this: once a csync signal is received (which is true if either the screen begins or the line begins), the firmware toggles the second signal (which is attached to "green") and generates the pixel for the text on screen...
1. drawback: the resolution of the output depends on the current horizontal sync rate
2. if the screen is "green" already at this area, no pixels are visible (so maybe output to all 3 colors [red,green,blue] produces better output).

keirf, 2019-07-15:
I have polled for interest in this feature and it's mixed to say the least. I may implement it in the future, it's quite a fun one to do. But the tight timing requirements, lack of available I/Os, and firmware Flash space, mean that I would do it as a separate firmware for STM32 "blue pill" board or similar. Perhaps emulating a 20x2 LCD, or a 128x32 or x64 OLED.

keirf, 2019-09-20:
This is now implemented in the new FF OSD project I'm hosting separately on github. Still to do is to support OSD and lcd/oled simultaneously, but that is coming soon.

closed 2019-09-20 by keirf