keirf/flashfloppy #252: Can't write on sy85

jpphillippe, 2019-07-17:
It can read disk images to SY85 no problem works great. but I can't for the life of me figure out how to save anything to the emulator/usbstick. Ive created a blank 720k hfe image (using HXC)  that is in slot 000 but every time i try to write anything to any slot it starts, looks like its writing then after 10-30 secs comes up "Error - File Not Found" seems like it write protected. checked the FF config and write protect is set to "No". 

Did a quick log using a upd from a previous bug report:

** FlashFloppy v3.3a for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Board: Standard
Config: Flash Slot 2 (ver 2, size 68)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: Invalid bus
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001070318109423
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
0:F: 'AMERICAN.hfe' 
Cache 101 items
Mode: Native
Current slot: 5/74
Name: 'AMERICAN' Type: hfe
Attr: 20 Clus: 00000007 Size: 2108416
Fast Seek: 1 frags
Cache 28 items
Cache 101 items
Before: After: Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416
Fast Seek: 1 frags
Cache 28 items
> USBH_USR_UnrecoveredError
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001070318109423
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
0:F: 'Animals.hfe' 
Cache 101 items
Mode: Native
Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416

** FlashFloppy v2.10d1 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Board: Standard
Config: Flash Slot 2 (ver 2, size 64)
Interface: Shugart (pin2=chg, pin34=rdy)
[!]Motor: Ignored
I2C: Invalid bus
Display: 3-Digit 7-Seg LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001070318109423
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
Config: Erased Slot 2
Config: Written to Flash Slot 3
0:F: 'Animals.hfe' 
Cache 94 items
Mode: Native
Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416
Fast Seek: 1 frags
Cache 29 items
New max: read_us=1993
New max: read_us=2497
New max: read_us=2550
New max: read_us=2551
New max: read_us=2590
New max: read_us=2609
Cache 94 items
Before: After: Current slot: 5/74
Name: 'AMERICAN' Type: hfe
Attr: 20 Clus: 00000007 Size: 2108416
Fast Seek: 1 frags
Cache 29 items
New max: read_us=2024
New max: read_us=4701
New max: read_us=4817
New max: read_us=4888
Cache 94 items
Before: After: Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416

Hope this is fixable.

keirf, 2019-07-17:
I assume you attempted a write during the logging above?

The image is not read-only, as that would be logged. There is no indication of a write occurring. Perhaps that signal is not getting through (that would be pin 14 of the SY85, going to pin 24 of the Gotek -- should go LOW (0V) during a write).

keirf, 2019-07-17:
Here is an up to date modified logging firmware for you to try: It will log whenever the state of the write-gate signal changes, and also logs its initial state whenever an image is mounted.

[ff_252_1.zip](https://github.com/keirf/FlashFloppy/files/3400533/ff_252_1.zip)

jpphillippe, 2019-07-17:
As far as I can tell yes. I changed disks with the log .upd. Tried to save all to disk 005. Got the error, then switched disks again to write the log. As you say it logged nothing about writing. Would incorrect jumper settings cause a problem?

keirf, 2019-07-17:
No, if it works for reads and disk changes then you can be pretty sure all available Gotek jumpers are set correctly.

jpphillippe, 2019-07-17:
I used "FF_Gotek-Logfile-v3.3d1.upd" I assume this was the correct .upd here's the new log

Fast Seek: 1 frags
Cache 28 items
WGATE: 1
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit
[11]
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001070318109423
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
0:F: 'Animals.hfe' 
Cache 101 items
Mode: Native
Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416
Fast Seek: 1 frags
Cache 28 items
WGATE: 1
Cache 101 items
Before: After: Current slot: 5/74
Name: 'AMERICAN' Type: hfe
Attr: 20 Clus: 00000007 Size: 2108416
Fast Seek: 1 frags
Cache 28 items
WGATE: 1
Cache 101 items
Before: After: Current slot: 6/74
Name: 'Animals' Type: hfe
Attr: 20 Clus: 00004491 Size: 2108416

Same procedure followed.
insert usb
switch to disk 006
switch back to 005
attempted a save all on ----new---02
started writing lcd showed 0-   0_   1-    0-   0_   1- 
then received "file not found" on SY85
swtiched to 006 to write the log.

You mentioned it should be 0v on pin24. is that as its writing? should I test with multimeter?

oguzog, 2019-07-17:
I tried unsubscribe from github even muted but still receiveing mails.

 Oguzhan Oguz

keirf, 2019-07-17:
Doesn't look like pin 24 ever goes low. I'm not sure you can reliably confirm that with a multimeter as it only ever will go low for short write periods (maybe only a few 10s milliseconds at a time).

jpphillippe, 2019-07-17:
ok, just as well my meters gone walkies. So you reckon dodgy connection? broken ribbon? or a bad gotek unit? rather than configuration/firmware?

keirf, 2019-07-17:
Bad hardware or for some reason the SY85 isn't even getting as far as trying to do the write.

You can test the Gotek I/O pins using the I/O Test firmware and a jumper wire: https://github.com/keirf/FlashFloppy/wiki/Testing-IO-Pins

keirf, 2019-07-17:
FINAL NOTE: Correct behaviour was found by using jumpers at JC and S1 while using this adapter board: https://www.ebay.co.uk/itm/Yamaha-SY-85-SY85-Floppy-Belt-Matsushita-EME-213YV-213-YV-Adapter-24-Pin/303147560808

closed 2019-07-17 by keirf

jpphillippe, 2019-07-17:
No idea why, but Setting the jumper on JC again fixed it so I'll close this now too. Cheers for your help!