keirf/flashfloppy #253: boot.ini Starting Proplem

koklica, 2019-07-17:
How can I start with boot. ini, how do I do? This is a problem

keirf, 2019-07-17:
You can't just put BOOT.INI on the USB stick. FlashFloppy only understands disk images, so you need, for example, to create a FAT12-formatted 720k or 12.44M IMG file, and put BOOT.INI into that image.

Mounting folders-as-FAT-images will come in a future release.

koklica, 2019-07-17:
I did.
img hfe did it but did not

keirf, 2019-07-17:
If FlashFloppy is mounting the image, and the host machine is at least able to read from the mounted image on the Gotek then there is no problem to be addressed at the Gotek/FlashFloppy level.

Only thing to suggest is that you make sure you have jumpers at locations S1 and JC at rear of Gotek if connecting to a PC.

koklica, 2019-07-17:
I can see the files inside when the program runs.
just doesn't see boot.ini at boot

keirf, 2019-07-17:
That is an issue out of scope for FlashFloppy I'm afraid. The Gotek is serving up the image correctly: setting up the filesystem contents of that image correctly for your attached host system is no different than setting up a real physical floppy disk. (ie: if you wrote your image to a real disk, and used it with a real floppy drive on your PC, it would behave no different).

closed 2019-07-17 by keirf

koklica, 2019-07-17:
real files on a floppy drive put into the home directory and the boot starts