keirf/flashfloppy #255: Hardware Mods wiki conflict (Motor-On + SD-Card PB15 reuse)

Volutar, 2019-08-06:
On Wiki page https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods#motor-signal it's said: 
**2. Connect Gotek interface pin 16 to STM32F105 pin 36 (PB15)**
And picture with wire soldered to pin 5 of U8 (which is indeed wired to PB15 of STM).

Lower, in "enhanced gotek/summary" section it's said: 
**PA15: Motor-On Signal**

So... Is motor-on PA15 or PB15?
Soldering picture insists it's PB15.
But:
**PB12-15,PC9 SD Card Connector (optional)**

So PB15 is used in 2 different enhanced gotek mods. How this signal can be used for SD card and motor-on signal? Something is wrong up there. Could you please clarify what's going on with PB15/PA15 thing?
Also, PA15 is not wired on STM at all.

Also another actually related thing.
I've just bought Gotek device. It happen to be model SFRC922D (with D) and there's an extra hole near power connector.
Picture of the PCB: http://volutar.eu5.org/sfrc922d-1.jpg
So jumpers are configured as 4+5 pins instead of "standard" 3+5. But interestingly, on the second picture of the wiki (https://github.com/keirf/FlashFloppy/wiki/assets/pheader.jpg) it is shown and named as SWDIO. I don't know where this SWDIO came from, but in my PCB revision this pin actually goes to PA13 of STM, so could be easily used (including motor-on).

I just wanted to use both mods, and found conflicting situation in wiki.

keirf, 2019-08-06:
If you advertise as an enhanced gotek on pins pc12-15 then motor is on pa15. Otherwise (eg standard gotek) it's on pb15.

Volutar, 2019-08-06:
Ah... got it... Wiki wasn't too clear about this, it looks to me more like a typo ;).

Okay. So, considering the fact, that PA15 is not wired anywhere, and only way to use SD mod with motor-on is to solder directly to STM leg 50. Right? It's... ultra-tricky...
In 922D revision PA13 is nicely wired. So why not to use PA13 signal in that particular mod combination?

keirf, 2019-08-06:
The 'enhanced gotek' mods were never particularly envisaged to apply to a std gotek. There are existing and upcoming third party boards which match the same form factor however

Volutar, 2019-08-07:
I couldn't find any third parties :(, so have to use this one (and wanted to make instruction manual). 

So the answer is "no"? That it's no way configuration gonna get extra setting which leg (pa12/pa15) is going to be used for motor-on?

keirf, 2019-08-07:
Don't these SD + motor enhancements already need you to direct solder to pins pc12-15? The soldering skill needed on a std gotek is already high (for enhanced gotek).

keirf, 2019-08-07:
Pa13 maybe. I need to check how easy in the code. I don't consider pb15 that hard to solder to however.

Volutar, 2019-08-07:
Yeah, you right. Soldering skills needed are already quite high (forgot about PC12-15). 
Nevertheless, PA13 could be used in any other gotek expansion with 922D revision (with pin config). Anyways, the answer was given about the topic, thank you:)

closed 2019-08-10 by Volutar