keirf/flashfloppy #256: Pc-ibm msdos : I can't get the gotek to work.

Starvak, 2019-08-07:
I have a gotek working perfectly on friend 500.
I wanted to use it on a PC, but I don't get it to work. (Amd k6-500 Win Me).

As I read, I have to do the following:
1- Connect the JC jumper to the gotek unit
2- I use the software to create a list of disks in usb memory.
3- Copy the FF.cfg and CONFIG.cfg files from the "examples" folder to the root directory of the pendrive.
The only thing I have achieved is to open unit A, very slow, and with file names with incompressible characters.
What am I doing wrong?

keirf, 2019-08-13:
Delete the cfg files, you don't need them. Press both buttons for a few seconds with no USB stick inserted, to reset the configuration cached inside the gotek.

Jumpers at locations JC and S1 only.

Use floppy IMG files which are exactly 720k or 1440k in size.

keirf, 2019-08-15:
Could be a bit like #257 ?

keirf, 2019-08-16:
Closing now. If there is still a problem, check if there are any floppy-related BIOS options to tweak. This helped in the case of #257

closed 2019-08-16 by keirf