keirf/flashfloppy #257: Not working F-F for 5.25 320k fdd

mrdavoudi1, 2019-08-12:
Hi
I have a 5.25 ffd on 386 pc that replace it with F-F on special PC for jumberca industrial machines.
The pc and the I-PC READ AND WRITE FLOPPY very well.
But when I connect F-F to the machine machine do nothing.
I use 3.32a firmware and check the log file with it related soft.
Do I have to put some changes for rpm or gap3 or mb/se speed.
Thanks

mrdavoudi1, 2019-08-13:
It means drive is selected,and red led also turned on  when I work on machine  but no changes in track number or side will take effect.
I change usb stick and  cable and emulator .
I prepare images with disk2fdi program and it says the 5.25 floppy is double sided with 40 track and 9 sector per track and 512 byte/sector.
Also the special dedicated IPC  that is 386 can't format  prepared 320k images but can read and write to them.
I also made images with WINIMAGE and the result is the same.
Regards

keirf, 2019-08-13:
It should just work - the default layout for 360k images is 40 cylinder, double sided, 9 512 byte sectors per track.

Did you put jumpers in the right place on the gotek? Jumpers at both  JC and S1 locations is usually correct for PCs.

mrdavoudi1, 2019-08-13:
I have mistake in my writing.
The images is 40 track  , "8 sec/track" and 512 byte per sector.
I test all the possible states.the drive work for first drive (S0) with non twisted cable and the green led (drive select) is turned on when drive is selcted.
Also change the interface with JC (open and close) and all the state in the FF.CFG.
5.25 drive have disk change on pin34(shugart interface).
I'm confuse.
Something is wrong.

keirf, 2019-08-13:
Ah yes well 320k IMG is automatically detected as double sided 40 cylinder and 8 sec per track. So should be fine.

If you want ready at pin 34 and dskchg at pin 2 then you don't need jumper JC.

mrdavoudi1, 2019-08-13:
With any one of up configurations the drive wasn't work.
No changes in track number and side number take effect.

keirf, 2019-08-13:
Does the gotek green led ever light? Have you tried the select jumper at each of s0, s1, and mo?

mrdavoudi1, 2019-08-13:
No.when I access the green led light.
Yes I test in s0 S1 not mo.

mrdavoudi1, 2019-08-13:
In s0 when i acess the floppy green led is light
In s1 when I acess the floppy green led is not light.
Also in similar machine like this with 360k floppy 5.25 F-F emulator work on s0 with jc open and no configuration else.

keirf, 2019-08-13:
Maybe the disk format is a little non standard on this system eg sector numbering? Skew? Etc

mrdavoudi1, 2019-08-13:
But I make  image from the disk sector by sector ad track by track and use that image.
Is there any parameter you suggest to change in FF.CFG  and IMG.CFG?

keirf, 2019-08-13:
Do you perhaps have a raw dump or HFE image of a real disk you converted?

mrdavoudi1, 2019-08-13:
I use this tool for many years.
http://www.oldskool.org/disk2fdi/
It first detect the disk properties and then image the disk.
And I have this image.and put this image on the stick.

keirf, 2019-08-13:
FDI images aren't supported by FlashFloppy. You need a headerless raw sector image (IMG).

mrdavoudi1, 2019-08-14:
Not fdi image.
The tool have many ability.
The images have  no header.
The tool save the bit by bit image as .IMG FILE.
how can i send you a .IMG file?

keirf, 2019-08-14:
Zip and attach to this ticket.

mrdavoudi1, 2019-08-14:
I don't know From where I can attach file.

keirf, 2019-08-14:
Accessing Github issues from a desktop web browser, you can "Attach files by dragging & dropping, selecting or pasting them."

mrdavoudi1, 2019-08-14:
[320dsk.zip](https://github.com/keirf/FlashFloppy/files/3500783/320dsk.zip)

keirf, 2019-08-14:
Okay, size is exactly 320k so that's fine. I assume you rename the images to have IMG file extension when you use them on FlashFloppy?

The contained filesystem is presumably an early MSDOS. I would guess maybe DOS 1.x as the usual 55AA identifier is not present at the end of the boot sector. It seems unusual to use such old disks with a 386 PC?

FlashFloppy v3.3a detects the format as C-H-S = 40-2-8, as you believe is correct.

Are you sure these disks are compatible with your 386 PC?

mrdavoudi1, 2019-08-14:
Hi
I change the extension myself.the original file is .img
The pc is 386 with special program to work with machine.
But the image is that one I sent to you
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

keirf, 2019-08-14:
Garbage vol label and file names?

keirf, 2019-08-15:
Could be a bit like #256 ?

mrdavoudi1, 2019-08-16:
Hi.
I set the computer setup parameter for drive a to "5.25 1.2m" and  Connect F-F  to position B of cable and software can work well with the emulator.🤔
But the image size is 320k
Then connect the emulator and use the image that software build on machine and it work as well as pc.
Thanks alot.

closed 2019-08-16 by mrdavoudi1

keirf, 2019-08-16:
What were the alternative options: 3.5/1.44m perhaps? It does seem weird... Glad it works now though!