keirf/flashfloppy #258: SFR1M44-U100 USB Flashing issue (With workaround)

jamhamster, 2019-08-14:
Hi, 

I got two Gotek SFR1M44-U100 VDisks for an Amiga and an Atari ST and flashed using USB via the included UPD file (FF_Gotek-v3.3a.dfu) but got nothing from the device, no display or signs of life at all. I flashed with flashfloppy-v2.13 and flashfloppy-v3.3a with the same result. I then put the Cortex firmware on it and this worked as expected confirming that the devices weren't dead. 

I had a lucky guess and generated the DFU file from the Hex file via DFU file manager and then I used this DFU file to flash the devices via USB as per the YouTube Video from TheRasteri and this now works as expected on my Amiga A500.

There's either something up with the UPD file in the Zip on this particular machine or the process above adds something 'magic' that was previously missing. 

I submitted this just in case someone else was having the same issue and also to thank you for such a great firmware.

keirf, 2019-08-14:
So that is using the Windows standard DFU update tool?

keirf, 2019-08-14:
Can you point me at exactly the tool you used (eg download link)?

jamhamster, 2019-08-14:
Sure, happy to help.  I used everything for flashing via USB in the recommended video (https://www.youtube.com/watch?v=yUOyZB9cro4) and the link to the tool I used is here: https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-stm32080.html

I used the standard Windows Driver and for some reason the DFU in the Flashfloppy downloaded zip didn't work until I created the DFU from the Hex which threw me for a bit.

If it helps, the actual devices I used is here:
https://www.amazon.co.uk/dp/B07Q4NQHW4?ref_=pe_3187911_248764861_302_E_DDE_dt_1

keirf, 2019-08-14:
I feel it must usually work as I have had no other reports. But perhaps I can find some interesting difference between the generated dfu files.

keirf, 2019-08-14:
Actually could you attach a dfu that worked for you, to this ticket?

jamhamster, 2019-08-14:
Sure, glad to help. I agree, I can't see any reason why it wouldn't work either and thought I was losing my marbles for a sec.  I have attached the firmware I created with the DFU tool from the FF_Gotek-v3.3a.hex. 

It could be that this is is a newer model and there's something 'hinky' about it?

[NEW_FF_Gotek-v3.3a.zip](https://github.com/keirf/FlashFloppy/files/3501474/NEW_FF_Gotek-v3.3a.zip)

jamhamster, 2019-08-14:
[image]

I have also attached an image of the drive with the pins in place where I programmed it in case this helps. I'm more than happy to close the post if you would like me to? it's purely as an FYI in case anyone is having the same issue. 

Again, thanks for creating such an excellent firmware. :-)

keirf, 2019-08-14:
Programming is really direct to the MCU and board can't have significant effect. And there is only one revision for STM32F105 ('Z'). May as well leave this ticket open a little while.

keirf, 2019-08-14:
I wonder if putting Cortex on the devices, and the method for doing that, somehow unlocked/unjammed the Goteks?

I tried the original v3.3a DFU image with the STM DFU tool (v3.0.6) and it worked first time with no issues at all (at least, I removed the programming straps, and the display worked!).

Diff'ing the original and new DFU images, the differences are in the Target Prefix and the DFU Suffix only:

Target Prefix: The NEW szTargetName field contains three nul-terminated strings instead of the expected single one. I think this is probably a bug in the STM DFU tool (eg. memcpy vs strcpy).

DFU Suffix: The NEW idProduct field is 0x0000 (vs 0xdf11 in the original DFU image). I don't believe this matters either way since at start of upgrade the tool warns that the device is in DFU mode and it cannot check whether the file is suitable for the device (which I assume would be based on checking the idVendor/idProduct fields).

keirf, 2019-08-14:
Arguably I should make the idProduct field 0x0000 to match the DFU tool.

By the way in original post you variously talk about the DFU file and the UPD file. These are different things! I can't imagine that confusion is your issue as the tool makes it hard to mistakenly use an UPD file, and would give an obvious error.

keirf, 2019-08-14:
Conclusion: I reckon the new DFU is a red herring and something else (eg even retrying the programming once or twice) somehow got things working, for some reason. I'm going to ask on the Facebook group if anyone else has had any problems, then close this ticket.

jamhamster, 2019-08-14:
Sure, do you have the name of the Facebook group? It'd save the group from having to go by second hand information. I'll be travelling for the next couple of days and will be away from my kit but I'll be happy to chime in where I can.

keirf, 2019-08-14:
Yeah, guess the name 😉

jamhamster, 2019-08-14:
Lol! I've requested to join. :-)

keirf, 2019-08-16:
I will close this now, but reopen if you gather any further interesting evidence. Thanks.

closed 2019-08-16 by keirf