keirf/flashfloppy #262: (not issue) Custom font

Volutar, 2019-09-06:
As far as I can see - F-F uses two particular embedded fonts (interesting, that source BDF files are unicode, but it allows ASCII-only symbols). Okay. But is it possible to add support of custom font files, so even 8x11 with display 128x32 would work in 3-line mode? or 8x8 for 4-line mode? Fonts which are perhaps be stylized as user wishes. 

There are plenty of old headerless dos fnt fonts, the format is simple - width is 8 pixels wide always, 256 symbols, so font file length is either 2048 for 8x8, or 3072 for 8x12, or 3584 for 8x14, or 4096 for 8x16 (all of them have been used in text mode), but other heights are also possible. Is it possible to make this? With char width and font file name taken from config.
Or something like that.

keirf, 2019-09-18:
It's a bit hardcoded at the moment for specific font sizes, but it could be generalised.

Volutar, 2019-09-18:
I know one font is embedded and hardcoded as "internal", which is used for writing startup and diagnostic messages. And another (6x13) is embedded and hardcoded also, but actually UNUSED, until particular setting on USB is read. But the font itself can be externalized as well, (like oled contrast, font, lines structure, so on), and some of users could also use system-specific look to add specific style, like ZX-Spectrum can use ZX font, Amiga users could use amiga font, and so on. 

I don't think font size wider than 8 pixels gonna be needed (but 9 pixels wide sometimes used, as 8 of font +1 of spacing). So simple bitmap set, 256*height*1 bytes long will do. If it's 6 pixels wide - it can be taken from the FF.cfg setting (rendering 8 pixels always, and moving N pixels to the right, how BDF actually works). 
[Fontraption](https://int10h.org/blog/2019/05/fontraption-vga-text-mode-font-editor/) DOS font editor comes with hundereds of fonts in this "raw" format. Nothing inside of files but font bytes. Char width is always 8, and height is taken from file extension (F08,F12,F14,F16 and so on). Perfectly works with DosBox on any platform.

Volutar, 2019-10-01:
Uh-oh. I meant not compile time (this would require full recompile/rebuild and reflash), but the RUN time. I’m talking about externalized custom font file on the flash card/usb thumb drive, near the FF.cfg. Sorry for inconvenience.

keirf, 2019-10-01:
Ah ok... Yeah still doable I guess

entonio, 2021-02-20:
I would very much like this but from the opposite direction - e.g. have a 16x32 font so that the mentioned 128x32 screen would allow a single line of 8 characters, or if it was 14x32 then 9 characters would fit. Or it could be 16x16 / 14x16, and using the 'd' option. I seem to recall the old DOS font was 8x8, so it could be duplicated to 16x16.
The main advantage of pixel screens when compared to 7-segment is having the image name, and for that one line is enough, so in my view it would be a fairly frequently chosen option.
I took a look at the codebase and I don't know that I'm comfortable with attempting a PR, at least not being sure the exact places that should be changed (for instance, I'm not sure how to generalise the `oled_convert_text_row_*` functions). And that's not even counting on finding an ff-idiomatic way to read the user-provided fonts - aiui, we'd have to provide an equivalent of `mk_font.py`, either to be run on PC or the gotek, but it would have to output something else than a C header if it's to be loaded at run time (not that I'd need that, but it's what this issue mentions).

HenrikErlandsson, 2023-01-05:
Agree with @entonio that the problem is text size on these small (but very sharp and high-contrast) OLEDs. You operate a rotary or buttons at arm's lengths and that's some justification for being able to see text at the same distance!

A custom 8x32 (or 16x32 fontsize with scrolling) could make the text 5mm/.25" in height and e.g. "CC.MyFileName|1B" would maximize the info shown (where CC. is cylinder + side, and | splits off last 2 chars in filename for multi-disk software, with maybe a CFG setting of content-row=4 (or 4d for 128x64 OLEDs) to generate this informative 16 char string to fit in the smaller width of the OLED vs the LCD 16x2.