keirf/flashfloppy #264: Missing write-back data near INDEX

SuperTurboZ, 2019-09-20:
In Ver.3.3a, with .HFE 600rpm/500Kbps, the title problem occurs in SMC-70 CP/M.

1.Images formatted with HxCFloppyEmulator can be read and written.
2.Formatting with SMC-70 will shorten the number of GAP1.
3.When writing to the image '2', the ID of the first track is lost.

https://twitter.com/SuperturboZ/status/1174636046790754306?s=20

It seems to occur only in the data immediately after the INDEX.

keirf, 2019-09-20:
600rpm / 500kbps - you mean this is a DD disk spinning at double rate?

Can you gather four hfe images: freshly formatted by hxc, freshly formatted by ff, and each of those two then written/modified by ff (presumably corrupting the data in case of the ff formatted hfe). Zip them up and attach them to this ticket.

keirf, 2019-09-20:
Also, does hxc formatted mean formatted by hxc firmware, or created/formatted by the hxc software that runs as a gui app on your pc?

SuperTurboZ, 2019-09-21:
>600rpm / 500kbps - you mean this is a DD disk spinning at double rate?
Yes.SMC-70 / 777 has a double speed 70 tracks 1DD.
Media is compatible with normal 1DD.Then access the FDC in 2HD mode.

The first .HFE image was generated with PC HxC software, but that is not a problem.
The problem occurs in the image after reformatting from SMC 70/777.

keirf, 2019-09-21:
Ok thanks that makes sense. Please send me the HFE images that I requested, when convenient, and I will take a closer look. I may get you then to run some tests, and/or run modified firmware.

SuperTurboZ, 2019-09-24:
In SHARP X1's 2HD mode, writing and reading of sector that the ID immediately after the INDEX are succeeded.
Also, writing and reading of sector that pass INDEX in the middle of the data block works well.
So FlashFloppy seems to have no problem.

I think it is necessary to reconfirm the interface problem between SMC-70/777 and GOTEK first.
Thank you.

keirf, 2019-10-01:
Please reopen if a FlashFloppy issue is confirmed.

closed 2019-10-01 by keirf