keirf/flashfloppy #265: Amiga Cannon Fodder save disk

bni, 2019-09-22:
It does not work to format and save in game. Im aware the save disk from TOESC etc does not work, since it is extended ADF. So I have created a HFE file like the troubleshooting FAQ says:

python scripts/mk_hfe.py save_blank.hfe

But when I try to format this disk in-game the drive light on the Amiga lights up green, the music continues playing, but after that nothing happens. I waited 5 minutes before resetting the Amiga. Have tried multiple times.

keirf, 2019-09-22:
Run Amiga Test Kit and run floppy drive write test to confirm writes are working okay.

Perhaps look for other ADF of Cannon Fodder? Possibly not all work well.

bni, 2019-09-23:
ok, just tried Amiga Test Kit adf

Writing of the 2 sectors worked, both on the Amiga Test Kit adf itself and on the hfe one I created

I tried these 3 versions from TOSEC:

Cannon Fodder (1993)(Virgin)(Disk 1 of 3)[cr FLT].zip

Cannon Fodder (1993)(Virgin)(Disk 1 of 3)[cr FLT][f load - save FLT].zip

Cannon Fodder (1993)(Virgin)(Disk 1 of 3)[cr CSL].zip

First 2 FLT ones, no difference. CSL one wouldn't load beyond disk 2 (looks like problem unrelated to this)

When I try to format inside the game, it will just continue with the music. If I after a few minutes change the disk it asked me "do you really want to destroy data on this disk?" changing the disk back and proceeding here got "error during format"

bni, 2019-09-23:
Don't have a external real floppy at hand sadly, otherwise I would have tried that

keirf, 2019-09-25:
Okay I will attempt to reproduce this. I assume the Format option is somewhere on the game's title screen or main menu, easy to find?

bni, 2019-09-25:
ok, thank you. Yes very easy to find on the title screen (The green hills with all the recruits) Its a floppy save icon in the upper right corner.

keirf, 2019-09-25:
What FlashFloppy version are you using, by the way?

bni, 2019-09-25:
v2.13 (Stable) using it with the Selector program and a Gotek with 3 digit display. Bought from Amigastore.eu a year ago or so.

keirf, 2019-09-25:
Okay and could you describe your Amiga setup a little, please. Model, Kickstart, expansions.

Also please run Amiga Test Kit (again) and do the floppy read test and let me know if it warns about anything at the start of the test.

keirf, 2019-09-25:
Ok, straightforward. I have a Rev 6 with 1.2 and 512kB expansion on my test bench.

keirf, 2019-09-25:
Ok I can reproduce this. I think perhaps it doesn't like a really properly blank HFE. I will investigate.

keirf, 2019-09-25:
Well it seems to work if you take an AmigaDOS disk image, in HFE format, and format it in-game. I have done this and now supply the pre-formatted Cannon Fodder HFE image. Please test it (it's linked in the Troubleshooting FAQ instead of the Python script!).

If it all works okay for you we'll close the ticket.

bni, 2019-09-25:
Thank you! I can confirm it works. Quite weird the game works like this but maybe everyone had pre-formatted floppies back in the day. THE FASTEST FORMAT... IN THE FUTURE ;-)

closed 2019-09-25 by bni

keirf, 2019-09-25:
It could possibly be an issue with the crack, as that will have bodged the disk routines including disk identity checks (which is where I think the unformatted hfe hangs). Anyway, problem solved so not worth digging deeper 😊👌