keirf/flashfloppy #269: Problems with display-order

5146jorge, 2019-10-03:
Hello, I am interested in changing the order of the information on the oled screen but I am unable to get it by executing the order : display-order = 1,0 
but I get nothing the information appears as always.

How can I get the file information to appear below the oled screen?

Greetings and a million thanks for your work!

keirf, 2019-10-04:
What Flashfloppy version? What oled size? Can you photograph or describe what is on each row, in order?

5146jorge, 2019-10-04:
Hi, I have the latest stable version my screen is 128x32 oled,the upper information is the name of the diskette and the lower one is directory, hint what I want is to reverse it, that is to say that in the lower part the name of the file appears and in the upper one the directory information appears, etc.

keirf, 2019-10-04:
`display-order` is a v3.x firmware feature, and it is tagged as such in the wiki.

So I recommend you upgrade to v3.4a -- it should be stable enough, but if you have issues you can always downgrade back to latest stable (same method as for upgrade).

closed 2019-10-04 by keirf