keirf/flashfloppy #270: Two sector bases for IMG.CFG RAW images

staticmem, 2019-10-05:
I'm now looking at accessing a RAW disk format that has different sector base on the first 2 cylinders on each side with the remaining cylinders all having another sector base.

For example:

Cylinders 0 - 1:
C00 H0: sectors 1-10
C00 H1: sectors 1-10
C01 H0: sectors 1-10
C01 H1: sectors 1-10

Cylinders 2 - 255:
C02 H0: sectors 21-30
C03 H1: sectors 21-30
...
C79 H0: sectors 21-30
C79 H1: sectors 21-30

I'm not sure if this is possible?, I would need to define two different sector base values as shown above.  Each side of the disk is identical.

keirf, 2019-10-05:
This was supported for a short while, for example see examples/IMG.CFG in v2.1a firmware.
It will be supported again in future, in a more flexible way, but I haven't got round to it yet.
Perhaps use v2.1a to access this feature, for now?

staticmem, 2019-10-05:
> # Optional: ID of first sector on each track (0-255[:0-255])
# Default is 1.
# Numbers may be expressed in hexadecimal with 0x prefix (eg. 0xab).
# Format x:y allows specifying different ID (x, y) for each head/disk-side.
# Format x specifies same ID (x) for both heads/sides of a double-sided disk.
# id = 1
>
I don't see how it can/should be declared.  Can you show me for above example I gave?

keirf, 2019-10-05:
Download and install v2.1a, and follow the instructions in its examples/IMG.CFG. For your situation:
```
id=1,21
```

Volutar, 2019-10-05:
Isn’t HFE supports of anything like this? I mean, it must be allowing lots of things, except for maybe fuzzy data.

staticmem, 2019-10-06:
> Isn’t HFE supports of anything like this? I mean, it must be allowing lots of things, except for maybe fuzzy data.

These are disk images that are Headless, there is no low level information in the disk image such as Sync GAP, etc. The RAW images just contain the data (i.e. 512 bytes) from each sector on each track.

@keirf:  I will try the v2.1a FW, hopefully today and report back.

Volutar, 2019-10-06:
> These are disk images that are Headless, there is no low level information in the disk image such as Sync GAP, etc.

Even when raw image file you're trying to use doesn't have necessary Sync/Gap info, this inter-sector data still spawned during device usage. So you either use low-level format with all necessary MFM and any sector sizes/numbers combinations for each track, by converting your images to HFE (which is already implemented), or ask for specific image format which is either way converted to HFE-like stream when accessed. So converting to HFE would be much more "long living" scenario.

keirf, 2019-10-06:
I don't know that the usual conversion software supports different sector ids for cyl0 vs cyl1+ but if it did you would be correct

staticmem, 2019-10-06:
I updated the FW with FF_Gotek-v2.10a.upd.

For this disk I have the following entry in IMG.CFG:

>[ds80]
cyls  = 80
heads = 2
secs  = 10
bps   = 512
id = 1,21
mode  = mfm

This is the command and response when running via the emulator that uses a Floppyio unit connected to the Gotek:

> ubee512 256tc fio-ds80-aa
disk_read: dsk_xread error: file= dsk_err=-29 Timed out waiting for response.
Track:   2   Side:   0   Sector:   2   IDside:   0
disk_iswrprot: dsk_drive_status error: file= dsk_err=-29 Timed out waiting for response.

It appears to have read C0H0, C0H1, C1H0, C1H1 ok (the system tracks) but fails once it reaches the data tracks that have sector numbers 21-30.

I can confirm that the DSK version of the same disk works and that this RAW image is good.

staticmem, 2019-10-07:
Some more digging...

Low level scans of the data tracks where sectors should be numbered 21-30 for DSK and RAW.

DS80 disk image RAW image format.

FLOPPYIO> scan a 0 2 2

```Drive: A  Cylinder:   2  Head: 0  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   2  Side: 0  Sector:   1  Bytes/sector:  512  crc1:  39 crc2:   7
Track:   2  Side: 0  Sector:   2  Bytes/sector:  512  crc1: 114 crc2:  84
Track:   2  Side: 0  Sector:   3  Bytes/sector:  512  crc1:  65 crc2: 101
Track:   2  Side: 0  Sector:   4  Bytes/sector:  512  crc1: 216 crc2: 242
Track:   2  Side: 0  Sector:   5  Bytes/sector:  512  crc1: 235 crc2: 195
Track:   2  Side: 0  Sector:   6  Bytes/sector:  512  crc1: 190 crc2: 144
Track:   2  Side: 0  Sector:   7  Bytes/sector:  512  crc1: 141 crc2: 161
Track:   2  Side: 0  Sector:   8  Bytes/sector:  512  crc1: 157 crc2: 159
Track:   2  Side: 0  Sector:   9  Bytes/sector:  512  crc1: 174 crc2: 174
Track:   2  Side: 0  Sector:  10  Bytes/sector:  512  crc1: 251 crc2: 253`
```
DS80 disk DSK image format.

FLOPPYIO> scan a 0 2 2

```Drive: A  Cylinder:   2  Head: 0  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   2  Side: 0  Sector:  21  Bytes/sector:  512  crc1: 232 crc2: 176
Track:   2  Side: 0  Sector:  22  Bytes/sector:  512  crc1: 189 crc2: 227
Track:   2  Side: 0  Sector:  23  Bytes/sector:  512  crc1: 142 crc2: 210
Track:   2  Side: 0  Sector:  24  Bytes/sector:  512  crc1: 158 crc2: 236
Track:   2  Side: 0  Sector:  25  Bytes/sector:  512  crc1: 173 crc2: 221
Track:   2  Side: 0  Sector:  26  Bytes/sector:  512  crc1: 248 crc2: 142
Track:   2  Side: 0  Sector:  27  Bytes/sector:  512  crc1: 203 crc2: 191
Track:   2  Side: 0  Sector:  28  Bytes/sector:  512  crc1:  82 crc2:  40
Track:   2  Side: 0  Sector:  29  Bytes/sector:  512  crc1:  97 crc2:  25
Track:   2  Side: 0  Sector:  30  Bytes/sector:  512  crc1:  52 crc2:  74`
```
As can be seen the RAW disk image is returning the sectors 1-10 instead of 21-30.

keirf, 2019-10-07:
You updated to the wrong firmware. Please use **v2.1a** not v2.10a.

staticmem, 2019-10-07:
Doh! ok I will try it again with the other version.

staticmem, 2019-10-08:
Updated,  It sort of works but if you refer to my original post the sector base of 21 needs to starts on cylinder 2 not cylinder 1.  This is what I see now with a scan:
```
Drive: A  Cylinder:   0  Head: 0  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   0  Side: 0  Sector:   1  Bytes/sector:  512  crc1: 202 crc2: 111
Track:   0  Side: 0  Sector:   2  Bytes/sector:  512  crc1: 159 crc2:  60
Track:   0  Side: 0  Sector:   3  Bytes/sector:  512  crc1: 172 crc2:  13
Track:   0  Side: 0  Sector:   4  Bytes/sector:  512  crc1:  53 crc2: 154
Track:   0  Side: 0  Sector:   5  Bytes/sector:  512  crc1:   6 crc2: 171
Track:   0  Side: 0  Sector:   6  Bytes/sector:  512  crc1:  83 crc2: 248
Track:   0  Side: 0  Sector:   7  Bytes/sector:  512  crc1:  96 crc2: 201
Track:   0  Side: 0  Sector:   8  Bytes/sector:  512  crc1: 112 crc2: 247
Track:   0  Side: 0  Sector:   9  Bytes/sector:  512  crc1:  67 crc2: 198
Track:   0  Side: 0  Sector:  10  Bytes/sector:  512  crc1:  22 crc2: 149

Drive: A  Cylinder:   0  Head: 1  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   0  Side: 1  Sector:   1  Bytes/sector:  512  crc1: 253 crc2:  95
Track:   0  Side: 1  Sector:   2  Bytes/sector:  512  crc1: 168 crc2:  12
Track:   0  Side: 1  Sector:   3  Bytes/sector:  512  crc1: 155 crc2:  61
Track:   0  Side: 1  Sector:   4  Bytes/sector:  512  crc1:   2 crc2: 170
Track:   0  Side: 1  Sector:   5  Bytes/sector:  512  crc1:  49 crc2: 155
Track:   0  Side: 1  Sector:   6  Bytes/sector:  512  crc1: 100 crc2: 200
Track:   0  Side: 1  Sector:   7  Bytes/sector:  512  crc1:  87 crc2: 249
Track:   0  Side: 1  Sector:   8  Bytes/sector:  512  crc1:  71 crc2: 199
Track:   0  Side: 1  Sector:   9  Bytes/sector:  512  crc1: 116 crc2: 246
Track:   0  Side: 1  Sector:  10  Bytes/sector:  512  crc1:  33 crc2: 165

Drive: A  Cylinder:   1  Head: 0  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   1  Side: 0  Sector:  21  Bytes/sector:  512  crc1: 115 crc2: 108
Track:   1  Side: 0  Sector:  22  Bytes/sector:  512  crc1:  38 crc2:  63
Track:   1  Side: 0  Sector:  23  Bytes/sector:  512  crc1:  21 crc2:  14
Track:   1  Side: 0  Sector:  24  Bytes/sector:  512  crc1:   5 crc2:  48
Track:   1  Side: 0  Sector:  25  Bytes/sector:  512  crc1:  54 crc2:   1
Track:   1  Side: 0  Sector:  26  Bytes/sector:  512  crc1:  99 crc2:  82
Track:   1  Side: 0  Sector:  27  Bytes/sector:  512  crc1:  80 crc2:  99
Track:   1  Side: 0  Sector:  28  Bytes/sector:  512  crc1: 201 crc2: 244
Track:   1  Side: 0  Sector:  29  Bytes/sector:  512  crc1: 250 crc2: 197
Track:   1  Side: 0  Sector:  30  Bytes/sector:  512  crc1: 175 crc2: 150

Drive: A  Cylinder:   1  Head: 1  Sectors:  10  Data rate kHz: 250  Encoding: mfm
---------------------------------------------------------------------------------
Track:   1  Side: 1  Sector:  21  Bytes/sector:  512  crc1:  68 crc2:  92
Track:   1  Side: 1  Sector:  22  Bytes/sector:  512  crc1:  17 crc2:  15
Track:   1  Side: 1  Sector:  23  Bytes/sector:  512  crc1:  34 crc2:  62
Track:   1  Side: 1  Sector:  24  Bytes/sector:  512  crc1:  50 crc2:   0
Track:   1  Side: 1  Sector:  25  Bytes/sector:  512  crc1:   1 crc2:  49
Track:   1  Side: 1  Sector:  26  Bytes/sector:  512  crc1:  84 crc2:  98
Track:   1  Side: 1  Sector:  27  Bytes/sector:  512  crc1: 103 crc2:  83
Track:   1  Side: 1  Sector:  28  Bytes/sector:  512  crc1: 254 crc2: 196
Track:   1  Side: 1  Sector:  29  Bytes/sector:  512  crc1: 205 crc2: 245
Track:   1  Side: 1  Sector:  30  Bytes/sector:  512  crc1: 152 crc2: 166
```
I know you want to make changes to make it more flexible but for my purposes I just need it to start on cylinder 2 for the other sector base (21) values.

Is it just a matter of poking in a different value into this v2.1a UPD image, if so what offset into the image?

keirf, 2019-10-08:
You'd probably need to pull the 2.1a sources, modify and rebuild. Can you do that?

staticmem, 2019-10-08:
Yes.  I've set up a build once before.  Can you give me a pointer to the source code and function I would need to change?

staticmem, 2019-10-08:
In src/image/img.c

```
static uint8_t sec_base(struct image *im)
{
//    unsigned int i = ((im->cur_track/2) == 0) ? 0 : 2;
//    return im->img.sec_base[i + (im->cur_track & (im->nr_sides - 1))];

// this works...
    unsigned int i = (im->cur_track < 4) ? 0 : 2;
    return im->img.sec_base[i + (im->cur_track & (im->nr_sides - 1))];
}
```

Works but probably breaks other things?

keirf, 2019-10-08:
Nope that looks correct to me. :)

staticmem, 2019-10-08:
I'll close this now.  If I have more issues I'll reopen it but hope you will find time at some stage to implement back in your more flexible code approach. Thanks for the help here :-)

closed 2019-10-08 by staticmem

staticmem, 2019-10-08:
Attached is the modified "v2.1a" compiled with a release of "v2.1z" in case anyone else needs it.

[FF_Gotek-v2.1z.upd.zip](https://github.com/keirf/FlashFloppy/files/3702709/FF_Gotek-v2.1z.upd.zip)