keirf/flashfloppy #271: write fault QD

kfolmo, 2019-10-16:
Using Roland MKS-100 write fails. I attach three files, the file labeled "original" is a file made by another Roland user on a different plattform.   It reads fine.  The file labeled "after write" is the same file after I have been trying to write a new sample on top.  The save process starts, then some time passes and the display on the sampler says "verify error".  After the write attempt the disk image is wrecked and will no longer load onto the sampler.   Also included the FFLOG.TXT.

[MKS-100 log.zip](https://github.com/keirf/FlashFloppy/files/3733048/MKS-100.log.zip)

kfolmo, 2019-10-16:
The Roland is probably as stupid as it possibly can be.  The sampler has 4 banks, labeled A-D.  You can make samples that only cover one bank (approx 1.second) or you can combine banks for longer sample time.  One bank uses up one side of a QD, so if you load a complex sample, then you need to read 4 QD sides... The reading of multisamples work fine, so a 4-sides acoustic piano loads very nicely just by turning the encoder to the next image.  When writing it first seems to read briefly through the entire disc to check if it contains anything.   If the disc has been used previosly it asks "Kill <samplename>" to make sure you don't overwrite anything.  When pressing save again, it tries to write the disc, which it somehow does to a certain extend, but it then tries to read it back and displays "verify error".

So, it can only read or write complete discs, I have searched through a couple of images with a hex editor, but it seems that it doesn't store the samples name in straight text.

keirf, 2019-10-16:
Well that sounds like it writes complete QD sides at a time. So something is definitely not right in that case.

I don't suppose you have a logic analyser? Long shot! ;)

kfolmo, 2019-10-16:
I'm afraid I don't have a logic analyzer, it's on the roadmap though...  Several have recommended til Intronix LA1034 so I have been thinking on that, but not really come there yet.

I just discovered that there is an "express save" that doesn't verify the disk before writing, but that didn't do any change for me.

keirf, 2019-10-16:
Regarding analysers, I can recommend the 8-channel Saleae clones on Ebay for fairly low-speed stuff like this. Doesn't help us right now, but they are only a few quid.

penfold42, 2019-10-16:
I can also recommend the LA1034 if you outgrow the the chinesium cheapies

keirf, 2019-10-17:
Well it's a bug, so I'll pull out the incredibly rare red bug label. ;)

I'm afk for the next week so I won't be looking at this myself until week of 28/10. I will make it a priority then, to try and repro this myself on the FF_TestBed.

keirf, 2019-10-30:
@kfolmo Hopefully that logic analyser won't be much longer. I've been working on Greaseweazle project flat out so haven't got back to this, and adding test support to FF_TestBed, yet.

keirf, 2019-11-05:
Okay here is a new firmware to try. I believe this issue is fixed by this new firmware, as I added tests to my TestBed and reproduced (and then fixed) the issue myself.

[ff_271_1.zip](https://github.com/keirf/FlashFloppy/files/3809127/ff_271_1.zip)

kfolmo, 2019-11-06:
Just tried it, it loads fine as before, but save is now both better and worse than before...  Now it gets stuck on "Save", it never gets to the verify operation, and it locks up totally so I have to turn the unit off.  BUT, when trying to load the disk image afterwards it reads the filename, and then gets an io error.  Earlier it did not write out the filename.  The logic analyzer is on its way but as always when you need something, that is the one package that is sent through both Ulan Bator and Timbuktu on its way.

keirf, 2019-11-06:
@kfolmo yes turns out I had more than one bug in the write path ;) I have now fixed another bug, and have a new firmware build for you to try:

[ff_271_2.zip](https://github.com/keirf/FlashFloppy/files/3816299/ff_271_2.zip)

kfolmo, 2019-11-06:
We are very very close!   I just made a sample, tried to save it, and it saved, proceeded to the verify and got a verify error.  BUT, did a power cycle and tried to load the image that was saved - and it worked, with correct name and as far as I feel, with the correct sample!!  I will try to do some more testing tomorrow, I'm not very familiar with the unit, so I need to make sure that I do things correct. regarding to the "A/B/C/D" structure of the sampler.

keirf, 2019-11-06:
@kfolmo when you do further tests, can you also try placing a jumper at position JC on the Gotek? I'm interested in whether there is any difference in how well it works with/without that jumper (it modifies behaviour slightly).

kfolmo, 2019-11-06:
yes, and I will also try to save some multi-disk samples to check if it manages to write to several images in a row.

keirf, 2019-11-06:
I have had reported success on S-10, including multi-disk samples. The tester happened to have JC installed from previous tests.

I note that S-10 has the same underlying board as MKS-100, so there is great hope :)

kfolmo, 2019-11-10:
After some testing today I can say that with the JC jumper set it performs brilliantly on the MKS-100.  I have to admit that it helped a little bit to read the MKS-100 manual - after all I've never used it with a QD since I got it the day before QD support by FF was launched.  Last test was to write and read a sample 4-sided sample which is the largest sample it can handle, and it worked perfectly!  Great work!

keirf, 2019-11-24:
This was fixed in latest release.

closed 2019-11-24 by keirf