keirf/flashfloppy #273: Follow on from Error 34 on Gotek display installed on an Atari ST #272

Coxside, 2019-10-18:
Following Kierf's advice, I installed the necessary files for the Signum application onto the USB stick,

This has indeed got rid of the Error 34 (thanks Kier), but the system is still unusable

The display on the Gotek now alternates between...

[Fonts] 
000/000        D:0
                    and... 
[]
000/000        D:1

FYI the files I've installed on the USB stick are as follows...

fonts folder, containg 36 fonts files, with the following files in the root directory outside the fonts folder...
                     AMOUNT.CLB  CV.SBK  CV.SDO  dcs9n.prg  dcs9n.rsc  dcs24n.prg  dcs24n.rsc
                     ENERGY.CLB  IMAGE_A.CFG   instal.prg   pr9n.par   pr9n.prg  pr9n.rsc
                     pr24n.par   pr24n.prg   pr24.rsc   scrop.acc  scrop.rsc  signum.par
                     signum2.prg  signum2.rsc  sigsep.dat   TEMP.CLB   TEST.SBK   TEST.SDO 

Any ideas/suggestions please?

TIA

Tony

keirf, 2019-10-18:
Flashfloppy works with disk image files on your USB drive. So those files of yours don't go direct onto your stick. You somehow put them in an ST image file and Flashfloppy mounts that. You might want to ask on Atari forums for advice on how to do that (tools, etc)

v-retro, 2019-10-30:
A Windows-based tool to transfer files to an Atari ST disk image is
available here:
http://www.atari.8bitchip.info/floimgd.php

closed 2019-11-06 by keirf