keirf/flashfloppy #274: Special firmware proposition - SIO/IEC device emulation

TzOk83, 2019-10-18:
If you have decided to add non-shugart floppy emulation, like QD, maybe you would consider implementing a special firmware for 8-bit Atari (SIO) and Commodore (IEC) computers?

penfold42, 2019-10-18:
1541 emulation would probably be beyond the goteks cpu capability.

Check out SD2IEC (simulated) and Pi1541 (cycle exact emulation) for the 64

keirf, 2019-10-19:
I reckon it could have a good go at 1541 but it is well out of scope for Flashfloppy and the pi1541 already exists.

TzOk83, 2019-10-19:
HxC also already exists ;) that's not an excuse. SIO is well within the scope, as there exists even a SIO2Arduino...

keirf, 2019-11-06:
Yeah but Pi1541 is free open source :)

And Atari is well served with SIO2SD... But it is more tempting. Is the SIO protocol documented?

TzOk83, 2019-11-14:
Yes, SIO protocol is documented:
http://ftp.pigwa.net/stuff/collections/nir_dary_cds/Tech%20Info/SIOSPECS.PDF
https://www.atarimax.com/jindroush.atari.org/asio.html

lgblgblgb, 2020-03-05:
Sorry to "invade" here, I've just meet the project, very nice :) My current thinking was to actually change the drive itself inside a Commodore 1541. So, I can have the "old feeling" of "real 1541" just the actual drive mechanics + heads + motor is replaced. However the connection between the 1541 board and the drive itself is not so much a "standard stuff" as I can imagine, so I wonder it is possible somehow to "hack in" a FlashFloppy'zed Gotek or something similar into a real Commodore drive (I guess it would also need GCR level image support then, or so?) keeping its original internals other than the drive itself as much as possible.

penfold42, 2020-03-05:
It’s possible - but a lot of work.
If I were doing it, I’d start with a 1541-II and build a daughter board under the U10 gate array to tap most of the signals needed

keirf, 2020-09-19:
Outside scope for FlashFloppy, and 8-bit Atari is no special interest to me. This will never get done.

closed 2020-09-19 by keirf