keirf/flashfloppy #276: FF 2.13: Sinclair QL: Format of IMG files failing without IMG.CFG file and HD/ED IMG.CFG tag configurations.

stephen-usher, 2019-10-19:
FF version 2.13.

I've been trying to get Sinclair QLs with various floppy disk interfaces to format images under Flashfloppy and have hit an issue with tagging.

Using a "Super Gold Card" interface and a floppy image of 80 track, double sided, high density size I get the following behaviour:

(1) No IMG.CFG file: Image fails to format, after system tries to detect floppy side and geometry it tries to format, for each track accessing side 0 briefly and side 1 for longer and then fails.
(2) Example IMG.CFG file: Image formats correctly as a 720K disk.
(3) Example IMG.CFG file with a tagged version of the default values: As with no IMG.CFG file.

You can see an example of what the system does when it fails here: [https://youtu.be/x5u96f2HR9A](https://youtu.be/x5u96f2HR9A)

Image file:
`-rwxrwxrwx  1 steve  staff  1474560 18 Oct 21:25 HD1.dshd80.img`

IMG.CFG entry:
```
[dshd80]
cyls = 80
heads = 2
secs = 18
bps = 512
```

keirf, 2019-10-19:
So in all cases you use that exact named 1474560 byte file?

1. Should work as there is a default geometry for 18 sectors per track for that image size. Are you sure there is no IMG.CFG on the USB drive?

2. Makes sense, but you mean it formats as a 1440k disk?

3. Makes sense as the defaults are for a 720k disk whereas your image is 1440k?

Or are you wanting to format it as 720k? I'm not clear.

stephen-usher, 2019-10-19:
The same image file for all tests.

I'm attempting to format it as a HD (1440K) disk.

In scenario (2) it seems that FF is using the default geometry (set in the IMG.CFG file) and presenting the file as a 720K floppy.

In scenario (3) I've set a tag entry for a HD floppy (18 sectors) and that's failing.

To note: a 720K image file fails to format as a 720K floppy without IMG.CFG existing and containing a default 720K disk entry and/or a tag entry for a 720K floppy.

keirf, 2019-10-19:
I am away for a week but will investigate myself then and check that Flashfloppy is doing the right thing. If you like put the exact IMG.CFG files you use for 2 and 3 in a zip and attach to this ticket. I can then inspect them and try them myself.

keirf, 2019-10-19:
Might you have a rogue host= line in FF.CFG? That could cause the case with no IMG.CFG to fail. Best put host=unspecified in FF.CFG

stephen-usher, 2019-10-19:
"host = unspecified" is  set. I'll zip up the files. The only difference between scenarios 2 and 3 is whether I comment out the [dshd80] tag configuration or not.

stephen-usher, 2019-10-19:
Zip file:
[ff-debug.zip](https://github.com/keirf/FlashFloppy/files/3746711/ff-debug.zip)

stephen-usher, 2019-10-19:
I've got further with the investigation, which only gives a potential solution for a sub-set of the problems on the QL but specifically formatting HD images.

The Miracle Systems Gold Card and Super Gold Card interfaces have a setting called "flp_jiggle" which when set, according to [this](https://superbasic-manual.readthedocs.io/en/latest/F/flp-jiggle.html) "forces the drive read/write head to make a number of rapid steps".

However, what I have noticed is that during the failed formatting the head is not returned to "Track Zero" before starting. If flp_jiggle is turned on then it does.

I'm therefore guessing that whoever wrote the code assumed that drives would home to track zero at some point during the command sequence and this assumption is wrong for Flashfloppy (and the Mitsubishi drives mentioned in the web page). Maybe other drives do seek there by default? I assume that Flashfloppy listens to the "Track Zero" line on the Shugart interface, I doubt it would work otherwise.

stephen-usher, 2019-10-20:
I believe I've discovered the issue.

In shugart mode pin 2 is set to disk change but the floppy interface is using it as density select.

There are obviously bugs in the Miracle Systems disk interface code as the format fails even explicitly giving the density when formatting. Also I have no idea why the "flp_jiggle" setting would fix this.

Maybe there needs to be a new host type, "ql", to set these defaults?

keirf, 2019-10-20:
Try pin02=dens in FF.CFG? That should do it. Assuming you really do want READY on pin 34.

You should feel free to tell me what to add to a Sinclair QL section in the Host Platforms wiki page.

stephen-usher, 2019-10-20:
Yes, after setting pin02 to dens the formatting works correctly.

I've tried both rdy and chg for pin34 and it makes no difference to the operation.

OK. I'll think about what to say later.

stephen-usher, 2019-10-21:
**Sinclair QL**
The Sinclair QL with any of the floppy interfaces tried so far (Sandy Q-Board, Miracle Systems Trump Card and Super Gold Card) will read and write IMG files without the need for an IMG.CFG file. The settings in the FF.CFG need to be set to the following otherwise formatting will not work and HD disks won't work on the Super Gold Card (even if FLP_DENSITY is used to manually set the disk type):

`interface = shugart`
`host = unspecified`
`pin02 = dens`
`pin34 = rdy`

The interfaces don't need index pulses or a track change delay so these don't need to be changed from the default.

It should be noted that all interfaces which use the default Q-Jump Toolkit II version of the FORMAT command fail to determine the correct number of sides on in a floppy image and hence format single sided, giving 720 sectors rather than the expected 1440. However, if a pre-formatted image generated by dumping a real floppy or from an emulator is used then everything works. It's only the FORMAT command which is broken. (The reason for this is being investigated.)

The version of FORMAT on the Miracle Systems Gold Card and Super Gold Card use a different algorithm and work correctly.

It is unknown if the FORMAT command in SMSQ/e running on a Super Gold Card works or not.

To create an empty DSDD image using the following command under UNIX/Linux:

`
dd if=/dev/zero of=myfloppy.img bs=512 count=1440
`

And an DSHD image using:

`
dd if=/dev/zero of=myhdfloppy.img bs=512 count=2880
`

An attempt has been made to support ED disk images however the data rate seems to be too high for the Gotek to reliably read/write, giving many CRC errors. Although formatting seems to succeed the image produced contains garbage.

stephen-usher, 2019-10-23:
P.S. I don't know if you have an IMG.CFG file whether you have to define all the image types. Can you merely add tags and leave the built-in disk geometries working?

keirf, 2019-10-23:
I think if you don't have a default section in IMG.CFG then it should fall back to the built in geometries. You will have to test to confirm.

keirf, 2019-11-07:
Okay I have added Sinclair QL to the Wiki. I paraphrased your original text in a few places. Please let me know if there are any problems.

https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#sinclair-ql

closed 2019-11-07 by keirf