keirf/flashfloppy #277: Enhancement: Add QL ED disk geometry to IMG system.

stephen-usher, 2019-10-19:
The Sinclair QL + Miracle Systems (Super)Gold Card combination has the ability to use ED disk drives. However, the geometry used is completely different from the IBM-PC one.

Could the following be added as a known IMG size/type?

cyls = 80
heads = 2
secs = 10
bps = 2048

keirf, 2019-10-20:
Does it work with IMG.CFG?

stephen-usher, 2019-10-20:
It did last night, but not when I just re-checked. I'm now getting CRC errors.

keirf, 2019-10-21:
May not be able to keep up with whole track formatting. Is it possible to try normal writes to an existing image?

stephen-usher, 2019-10-21:
Unfortunately I don't have an existing image and none of the emulators support ED format floppy images.

I think that I may have to call ED floppy disks unsupported/unsupportable. Obviously the data rate is too high. (On the original drives it was above the manufactuer's specification. PC disks were 80 track, 2 sides, 36 x 512byte sectors per track.)

stephen-usher, 2019-10-22:
I've been informed about a utility which will generate ED floppy images, so I'm re-opening and will test this evening.

stephen-usher, 2019-10-22:
I've placed a pre-formatted ED image file onto the USB drive and set up IMG.CFG and the drive works correctly for reading and writing normal files, so it does look to be that formatting is a step too far and the Gotek can't keep up.

keirf, 2019-10-23:
Would you like to provide updated wiki text for ED support?

Also I am happy to provide you a firmware to test which may support ED formatting too. Perhaps next week?

keirf, 2019-10-23:
Actually your stick is really slow at small writes and won't keep up. Strange as it is a decent branded stick. Perhaps try another? You are looking for significantly shorter sector write times in FFLOG.TXT. Note that currently the timing is highly variable and never less than about 12ms,often way more. Whereas the host is pushing out a sector every 5ms.

stephen-usher, 2019-10-23:
I've updated the Wiki entry text in the other issue thread.

I think it's probably safer to manage expectations and state that it doesn't work. If people find that it does for them then that a bonus but it's not something you'll have to support.

keirf, 2019-10-23:
I was wrong in my 5ms per sector. Because these are 2k sectors the host pushes one out about every 20ms. So your stick is close to coping, it just has slow writes every now and then. I'm confident you could find a stick that does keep up.

stephen-usher, 2019-10-23:
I wonder if it depends upon whether a block has already been used and hence needs copying, blanking and rewriting. I very much doubt that USB sticks have wear levelling (and definitely no TRIM).

Anyway, if it's not dependable then it's probably best not to advertise that it may work.

Volutar, 2019-10-24:
Isn't it possible to cache the data which gotek is going to write, in order not to overload flash cards/usb sticks? And actually inform if it gets overload?

stephen-usher, 2019-10-24:
@penfold42 Doesn't the Gotek use a PIC18F4620 like the HxC devices? If so then that has 3,968KB of SRAM on-board. Of course, that memory would also hold the Flashfloppy system.

keirf, 2019-10-24:
64k sram. Currently 32k is used to buffer incoming raw bitcells. Enough for an HD track but not a whole ED track (50k). Sectors are written to usb as they are decoded.

There is also a buffer cache on top of USB which can cache approx 16k. This is underutilised on the write path (data doesn't pass through it at all). On the write path we should probably greedily decode into the buffer cache and then write back dirty usb buffers. We should be greedy because that 50k raw bitcells decodes to 18-20k of sector data. So it's much more efficient to cache that.

This is what I would aim to do if I work further on this ticket.

Volutar, 2019-10-24:
> From memory, the stm32 cpu only has 32k ram in total.
> There probably isn’t a lot left to implement a write cache.

Well. The question consisted of two parts - caching (as much as possible) and informing about writing problems. F-F suffers from that infamous silent “didn’t work out”, when you find writing gone bad only after everything else checked. It’s not how it expected to be. Proper feedback about writing failure is really needed.

keirf, 2019-10-24:
The issue with this SanDisk stick however is that I see a worst case sector write back time of 140ms. During that time we cannot decode more sectors (as that is done in the main loop same as write back). We would instead buffer in the 32k ring which can buffer 64% of an ED track. But in 140ms we receive 70% of an ED track. That's no good we need a less choppy usb stick. And if we have that, the 32k ring may suffice alone.

Volutar, 2019-10-24:
Is it possible to add some sort of easy to use benchmarking, to test writing time and inform if its not sufficient for proper writing?
I still don't know if i am using good enough (though didn’t really needed to write, yet).

keirf, 2019-10-24:
Further thought on this: if I allowed dirty buffers to accumulate in the cache and write back a bit more lazily. I could do this only for tracks that don't fit in the 32k ring buffer and only when I detect a whole track write.

I think this would make ED formatting reliable.

keirf, 2019-10-24:
@Volutar At HD rates and lower, whole track writes are already reliable as a whole track can be cached in the ring buffer. Furthermore write to the next track is delayed until writeback is complete and FF starts generating index pulses again, so the host cannot get ahead of us. In short I wouldn't worry about it. This ED rate formatting problem is fairly specific.

PeterBortas, 2019-10-24:
Blue sky idea: If the ring buffer can handle the overhead of some very simple RLE-compression that should cover the ED formatting without problems, right?

keirf, 2019-10-24:
Compression of the ring is possible but probably more hassle than the dirty buffer scheme.

stephen-usher, 2019-10-24:
Then again, seeing the biggest problem is during formatting where most of the data being written will either be zeros or a long run of another value this should compress massively even with a simple RLE encoding.

Volutar, 2019-10-24:
@keirf i see. 
Random thought about compression. Ring buffer is MFM based (ie 16 bits per 8bit). But it’s not entirely aligned. Thought I think compression via MFM encoding-decoding is possible. ED disks if i’m not mistaken have 25000 bytes per track (or 50000 bytes in MFM form). 
There is an UDI image format used in zx emulators. It’s pretty rawish, but it stores aligned byte stream (not MFM) with supporting bitmask of special sync bits (for C2 and A1). So its like 8+1bit. Maybe its not bad idea to arrange memory buffer similar way (to avoid MFM 2x sized stream).

stephen-usher, 2019-10-25:
Actually, if you're buffering/caching a whole track then maybe you can support some silly tricks drivers try to do, such as formatting more sectors than exist (assuming that it'll wrap around to the beginning of the track again) by just resetting the pointer in the ring buffer to the beginning if there's an overflow of data, so simulating what the dumb hardware would do? Just a thought.

keirf, 2019-10-25:
Those tricks already work fine with HFE image format. The problem is that IMG is in a sense pre-formatted. The sector arrangement and numbering is implicit (because IMG images are headerless and don't hold such information explicitly). Once the track wrote gets written back into the IMG and reading starts up again, everything goes back to the same fixed sector numbering...

In short if you need tricks you must use HFE.

stephen-usher, 2019-10-25:
Yeah, I guess it would only work if you recorded the raw MFM data stream and reset the recording to the beginning of the buffer every 0.2 of a second after the "index mark" and then interpreted the data after the write had finished. Then you'd have to worry about the overwritten "track" not being on proper sector boundaries. Fair enough.

Volutar, 2019-10-25:
> From memory, the stm32 cpu only has 32k ram in total.
> There probably isn’t a lot left to implement a write cache.
Datasheet says stm32f105has 64kbytes of sram.

keirf, 2021-04-25:
If there is anyone still watching this ticket, I'd like to say I'm hopeful that ED formatting will work as of FF v3.25. The original issue was not one of buffering (ring overflow) but a bug in write ring handling which meant that ED track formatting could *never* have worked in earlier versions, even with a sufficiently fast USB stick.

I am interested if anyone would care to test ED formatting on v3.25?

slartibardfast, 2022-05-12:
hi not sure if IBM PC ED formatting is of interest? 

still have trouble with lengthy writes & bad crc when formatting 2.88MB / ED disks on a NuXT 2.0 & latest FlashFloppy 4.4a

perhaps it's just usb drive not able to keep up, but i've tried several devices.
let me know if there's anything at all i can do to test options or anything else!

thanks very much!
-dc

[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/8677245/FFLOG.TXT)
[DDED.zip](https://github.com/keirf/flashfloppy/files/8677253/DDED.zip)

cyberluke, 2022-07-06:
Hello, I’m back. I can test it on new version. I think no need for backporting to v3.25 if userbase is not requesting it.

keirf, 2022-07-06:
The other fix here would be to obtain one of the new Goteks with AT32F435 MCU. These have lots of RAM, relatively speaking,  for buffering ED writes.

keirf, 2022-12-19:
Use a 435 based Gotek is the answer here. Closing.

closed 2022-12-19 by keirf