keirf/flashfloppy #278: No disk in drive "A" on ibm-pc

kooroshi60, 2019-10-19:
I flashed my Gotek with FlashFloppy and tested its function with io-test firmware. Every thing was OK.
Gets detected with these Jumper settings in Windows.
Straight cable : S0 and JC
Twisted cable: S1 and JC
This is the issue I am facing with:
Buzzer makes noise when trying to access the disk images and on the LCD i can see it is stuck in T:01.1/T:00.0 (flattering around these two values) and the PC doesn't show the image contents. after some 4 seconds of trying, Windows says " Please insert a disk to drive A". 
Floppy image sizes are exactly 1440 and 740 Kbytes but I don't know why it keeps saying that no floppy is inserted. 
I Ejected and reinserted the images but it didn't solve the issue!
I tried 4 different USB sticks and many different floppy images but nothing changed!
The original GOTEK used to work with straight cable and S1,JC jumpers strapped.
I was reading the #256 and #257 but didn't figure out what is wrong with my GOTEK.

keirf, 2019-10-20:
You mean 720k and 1440k images?

Does your ribbon cable have a twist in it (it should do)?

Bad ribbon cable maybe?

Did you test all gotek pins with iotest?

Flashfloppy version?

kooroshi60, 2019-10-20:
Thank you for your swift response.
Yes 720Kb. Sorry it was a typing mistake!
Yes I am using a ribbon cable with a twist in it. Cable is OK as continuity test confirms it but anyway I'll buy another ribbon tomorrow and test it again.

I also tested the FF with straight ribbon which worsens the situation. In this case, PC does not hesitate AT ALL to show "Please insert a disk to drive A" message. As if the USB stick is not inserted into GOTEK.

IO-Test passed cleanly and all pins,buttons, encoder where functional as expected.

First I flashed the 2.13 which is the stable version and then I upgraded to 3.6a ; In both cases the result is same.

kooroshi60, 2019-10-20:
IT IS UP AND RUNNING NOW. The problem was a little piece of solder which was stuck between the MCU pins. I cleaned it and now it works like a charm.
Jumpers S1 and JC are strapped.
Thank you for your help.

closed 2019-10-20 by kooroshi60