keirf/flashfloppy #279: FF 2.13: Sinclair QL + Double Density only interfaces fail to detect double sided image.

stephen-usher, 2019-10-20:
Sinclair QL computers using earlier disk interfaces, such as the Sandy Q-Board and Miracle Systems Trump card attempt to determine if the drive is single or double sided by trying to format track 0 sides 0 and 1 and then attempting to read certain sectors.

However, this causes FF to error, e.g. (see attached log file)

> Write 8[09]/9... 6039 us
IMG Bad Sector Offset: 4294961184 -> 9
Write 0[01]/9... 3512 us
Write 1[02]/9... 5982 us
Write 2[03]/9... 7468 us
Write 3[04]/9... 7479 us
Write 4[05]/9... 7630 us
Write 5[06]/9... 7363 us
Write 6[07]/9... 6424 us
Write 7[08]/9... 7478 us
Write 8[09]/9... 8282 us

The interface takes this as an indication that there is no side 1 and formats the drive successfully as a single-sided disk image.

Is there a way that FF can mitigate this error and allow the system to format correctly?

[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/3747811/FFLOG.TXT)

keirf, 2019-10-20:
I think that error is spurious. It happens because the write doesn't start at a sector boundary. But you can see 9 sectors are written successfully. So there was no error writing all sectors in the track?

stephen-usher, 2019-10-20:
Unfortunately the test for how many sides the floppy has is opaque. Once it decides upon it being singe sided then the format works perfectly, giving half the capacity.

keirf, 2019-10-21:
Well the log file overflowed so it's missing the start of the format. Try ejecting the disk image soon after it has decided single sided, and attach the FFLOG.TXT then

stephen-usher, 2019-10-21:
I'm not sure if this helps at all.
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/3752578/FFLOG.TXT)

keirf, 2019-10-23:
It looks like format writes extra sectors to track 0 side 1. These will not fit and will overwrite the normal sectors. It will then try to read normal sectors on side 0. If it cannot then side 1 and 0 are the same (single sided).

This kind of trick is not a fit for FF IMG handling. Use HFE and then convert to IMG after the format. You could even keep that as your formatted blank, clone and relabel it whenever you need a blank?

stephen-usher, 2019-10-23:
Ah, that would make sense. Pity. Oh well.

I don't know how critical CPU time is in this case but would it be a possibility to use the requested sector number modulus the number of sectors if and only if it's greater than the number of sectors. In the best case this is a compare and test, in the worst case a compare, test, integer divide, multiplication and subtraction. It would more closely simulate the hardware it seems.

keirf, 2019-10-23:
Host doesn't "request a sector" from the floppy drive. The drive
continually returns the track contents which the host (floppy controller)
picks apart to find the sector of interest.

stephen-usher, 2019-10-23:
Fair enough.
Thank-you for examining this.
I can re-write the stuff for the Wiki I posted on the other issue thread.

closed 2019-10-23 by stephen-usher