keirf/flashfloppy #280: E34 with LED, oled and OSD both work

penfold42, 2019-10-25:
Follow on from #272 

It is weird ...
four goteks give me E34 with the 3 digit led display but work fine with oled or OSD.

Have checked with latest firmware and first noticed it about a month ago flashing a friends.

Display is set to auto.

keirf, 2019-10-25:
Some Goteks do this and some don't? They stop it when you install a different display type?

What do you use to power the gotek for these tests?

Firmware version? USB stick brand and model? 

Can you run a debug firmware and collect serial logs?

penfold42, 2019-10-25:
All the goteks I have tried do it.
Power up with LED - E34
Power off, change to OLED or OSD and power up and they work fine.

Powered by either my Amiga with PC PSU or from the 5v line on an FTDI breakout.

Firmware - built from master 2 days ago and serial flashed as well as using a 3.4.upd file.
They do it with 3 different USB sticks.

How do I enable a debug build ? 
Happy to grab serial logs.

Personally I don’t care for the led, but given the weirdness and a couple of other reports, happy to explore it.

penfold42, 2019-10-26:
Sorry - I suspect It’s user error:

Subfolders are accessible only when using an LCD/OLED display.

closed 2019-10-26 by penfold42