keirf/flashfloppy #281: Kawaii Q80EX working perfectly!

Warrenverc, 2019-10-28:
After a week of trail and error I finally found the solution to getting the Kawaii Q80EX working perfectly with the Gotek USB floppy emulator.
1. Jumper settings are as follows : 
JA = CLOSED
S0 = CLOSED
MO = CLOSED 

2. Connect the Gotek to the Q80EX making sure that the ribbon cable is upside down, or the Q80EX won't turn on. 

3. Flash the latest firmware update and bootloader. 

4. Make an image file from an original Q80EX floppy. 

5. Copy the image with a few more copies naming them DSKA0000. img, DSKA0000. IMG AND SO On.... 

6. Copy the file HXCSDFE. CFG to the root of the USB flash drive along with the disk images. 

7. Put the flash drive into the Gotek device and power up the Q80EX. 

That's it! 

Works great!

Warrenverc, 2019-10-29:
[Q80EX_HXC.zip](https://github.com/keirf/FlashFloppy/files/3781128/Q80EX_HXC.zip)
Another quick note is that the HXCSDFE. CFG in the flash floppy distribution doesn't work. 

I have included a ZIP FILE with 15 empty Q80EX floppy images that are empty. 
Also version 1.0 of the HXCSDFE. CFG that works fantastically. 

Enjoy....

Warrenverc, 2019-10-29:
I have now found out that it works without HxC_Compat_Mode also.
Much easier and in the FlashFloppy native mode.
With the LCD/OLED: FlashFloppy allows you to select between any valid image file and any folder within the current directory. 
Selecting a folder will make that the new current directory, and list any image files and subfolders within.
You can name folders as you wish and the floppy images do not have to be consecutive number naming but you can name any disk with more descriptive names like "U2SONGS.img"

No need for an FF.CFG or IMG.CFG.

keirf, 2019-10-29:
Do you mean JA closed? Perhaps you mean JC?

Warrenverc, 2019-10-29:
No JC doesn't work JA does and really well. It loads and saves within
seconds!

It's totally fully reliable and I am so happy!

Regards
Warren

Warrenverc, 2019-10-29:
> Can you send me a photo of your jumper config? You see, JA is the header
> for a third button (Select), and if you jumper it, it's like that button is
> permanently pressed. That should cause you to be stuck in the Firmware
> Update bootloader. So I can't put your instructions into the wiki as is. I
> need to understand first exactly how/why it works for you!

Will do ! I read somewhere about the Yamaha synth with the same
configuration and after almost giving up I tried it and it works!

chrispoacher, 2019-10-29:
I know my Q80 works fine with just S0. Not sure why the EX is so different?

Warrenverc, 2019-10-30:
Here's the photo of my jumper settings and the link to the site I found
about the Yamaha synth: -
https://ivanmonterosso.wordpress.com/2014/10/05/the-gotek-floppy-emulator-the-right-way/

> .
>

Warrenverc, 2019-10-30:
The photo 
[image]

Warrenverc, 2019-10-30:
There is a huge difference between the Q80 and the Q80EX drive configuration, I have read.
I visited many sites even those from HXC and people with the Q80EX didn't get the emulator working with the Q80 settings. This solution works really fantastic for me!

Warrenverc, 2019-10-30:
One more thing if the JA jumper is not closed then the Q80EX shows up on the screen
"DISK UNAVAILABLE".

keirf, 2019-10-30:
Okay can you also send a photo of the OLED display when no USB stick is inserted. (ie. banner screen). Thanks!

Warrenverc, 2019-10-30:
Here are the pictures of the the original drive with the brand details Teac FD 235F and the jumper settings RY DC and D0 are closed. Also pictures of the Gotek without the USB stick and one with the USB stick. 
[image]
[image]
![IMG_20191030_161409](https://user-
[image]
images.githubusercontent.com/57118107/67872636-8b914200-fb32-11e9-88e5-8c3407800b15.jpg)

keirf, 2019-10-31:
Here's what happens when I place a jumper at JA when running v2.13. As expected it is stuck in the bootloader. Anyway, I am very pleased FlashFloppy is running on your EX :)

[image]

closed 2019-10-31 by keirf