keirf/flashfloppy #282: Different QuickDisk Pinouts

steeph-k, 2019-11-09:
I would like to use a Gotek in a Triumph Adler word processor because its QuickDisk drive is broken. The wiki has a pinout for a 10 pin connector. However devices with integrated QuickDisk drives use many different connectors and pinouts. Most of them seem to use all 12 pins. This should probably be mentioned in the wiki.
I could find pinouts for a handful of connectors. But not the one my device uses. Before I try every possible combination and brake something. Is there a way of detecting which pin is used for what?
Pins 1 and 12 are both connected to ground when the device is turned off and I have 5 V at pins 2, 10 and 11 after turning it on. (Pin 6 has 4.95 V)

keirf, 2019-11-09:
Google "quickdisk 12 pin" and you should find the 12 pin mapping for Famicom Disk System which could be the same? The extra pins are Battery Sense and Motor Sense which perhaps can be ignored?

keirf, 2019-11-09:
Actually your ground pins don't sound a good match up with the FDS mapping. You could at least work out which pins are connected direct to 5v and which are ground? The rest a mismatch is less critical. If you could put a logic analyser on the other pins you might get a better idea which does what.

steeph-k, 2019-11-09:
Thanks! I'll do that. But I'm not sure I'll be able to identify pins from that data. So I guess I'll come back, whether I find my answer or not.

BTW: The FDS connector pinouts can be found here: https://hxc2001.com/download/datasheet/floppy/thirdparty/QD/VirtuaQD_Manual.pdf

The one for the drive should be this (I'm guessing):

| 1 | GND |
| 2 | N.C. |
| 3 | WRITE DATA |
| 4 | READ DATA |
| 5 | WRITE GATE |
| 6 | MOTOR ON |
| 7 | QD RESET |
| 8 | WRITE PROTECT |
| 9 | DRIVE READY |
| 10 | MEDIA SENSE |
| 11 | BATTERY SENSE |
| 12 | VCC |

keirf, 2019-11-09:
Depending how highly integrated the circuitry is to non standard asics, perhaps you can trace QD header pins back to chip pins and work out some functions that way, or eg which are inputs and which outputs

steeph-k, 2019-11-09:
If I knew what the procedure of reading or writing was I might be able to connect one pin after another so I'd only have to guess one at a time. One pin drops low when trying to read or write a document. I guessed that might be Motor On. But I don't know how to proceed from there. Does it expect a ready signal from the drive?

I'll look at the PCB too.

keirf, 2019-11-09:
Yes pin going low on read and write will be motor. Unless you have a scope or analyser the more dynamically changing signals will be hard for you to determine. If you can get traces from your original drive we can work out your signals one by one.

steeph-k, 2019-11-09:
I do have a logic analyser. It's just that nothing else changed state after or before the one pin dropping.

So I've had a look at the board in the drive. Thanks for the idea! I can't find the data sheet for the large IC, 3204M. This would probably be enough to identify most pins and guess the others.

[image]
[image]

steeph-k, 2019-11-09:
I found this: https://torlus.com/floppy/forum/download/file.php?id=1596
It fits the pins I knew so far. So I've connected it. It is the 10 pin connection after all. Pins 11 and 12 are just a separate 5 V line for the motor and another ground. Now I do get an error message if no disk/image is inserted. But if there is one, still nothing happens. Pin 10 drops low, nothing after that.
Nothing was written to the blank image. It's the same when trying to read.

keirf, 2019-11-09:
I wonder if pin 9 (WG) is opposite sense to what Flashfloppy expects. On the 10 pin drives it goes HIGH to indicate a write. But here as it is /WG it may go LOW. I can do you a test firmware....

keirf, 2019-11-09:
Try this

[ff_282_1.zip](https://github.com/keirf/FlashFloppy/files/3827856/ff_282_1.zip)

steeph-k, 2019-11-10:
Thanks. This didn't change it though. It wouldn't affect reading also, as I understand. I'll have another look at the code later to see if I can make out what *should* happen next.

keirf, 2019-11-10:
Hmm yes that's true I think. The spurious write request would be ignored and you should see read data.

You won't see /READY go LOW unless you have: an image mounted, /MOTOR driven LOW, and /RESET driven HIGH.

keirf, 2019-11-11:
Apologies, I didn't realise I hadn't documented the Gotek jumper configuration in the wiki!

You should be testing with a jumper at location **MO** only. Locations S0,S1 are unused.

This may get you further ;)

steeph-k, 2019-11-11:
Aw, I should have asked about that earlier. I'm travelling till the weekend. But I'll try that as soon as I can. Thanks!

steeph-k, 2019-11-22:
I'm getting nowhere.

Everything I try stops after a while with the message "Disk damaged or not initialised." Even trying to initialise the disk gives me the same message.

This is how initialising looks with the firmware with inverted WG:
[image]
With the firmware from the repo I don't get the blocks on RD and the data on WD continues for much longer. A good sign? It looks like this all the way through:
[image]

And this is trying to write a document (trying to load one looks the same):
[image]
This is with WG inverted again.

steeph-k, 2019-11-22:
This is initialising with the firmware from the repo:
[image]
[image]
[image]
[image]

keirf, 2019-11-24:
Okay the release/repo firmware is definitely better. The write extending forever is definitely not good of course ;) I think the host is waiting for /READY to deassert, but for some reason that never happens. A bug? I think so ;) I will have to investigate this and get back to you with another firmware for test!

 Thanks!

keirf, 2019-11-24:
Before I investigate: Was your test where the write extended forever done with latest release firmware (v3.7a)? This included some big fixes for QD write handling, so you should try that first if you haven't already.

steeph-k, 2019-11-24:
Oh, I missed that. I'll try it in a minute.

steeph-k, 2019-11-24:
Well, it's different. Looks better, but it's not quite working yet. I'll post a screenshot soon.

steeph-k, 2019-11-24:
So, trying to load or write a document looks still the same. 5 relatively long blocks on RD, then it stops with the error message. I'm guessing it's retrying a few times before it gives up.

When initialising I get this:
[image]
And it's also five blocks on RD overall.

steeph-k, 2019-11-24:
The long writing part:
[image]

The short writing part:
[image]

keirf, 2019-11-24:
Did that test run start with a format/initialise? Looks like it; then straight into a read/verify attempt?

Can you measure the width of low pulses on write data, and time(s) between successive falling edges?

keirf, 2019-11-24:
Actually looks like wdata here is active high... In which case measure high pulse width and rise-to-rise. May also be relevant that I expect active low wdata and perhaps that does break things here

steeph-k, 2019-11-24:
The second, short writing part is just a 100kHz (10µs) pulse. It stays high for 1.75µs.

The first, longer part has a few patterns I can make out. So it looks like real data. Do you want a data export?

I'm not sure what you mean by successive failing.

BTW, the disk image is full of stuff now. Mostly Us and Is, some *s.

keirf, 2019-11-24:
Sure I use the saleae software myself so can load into that

steeph-k, 2019-11-24:
Good, I skip the 4 GB CSV export then. :)
Here is the saved capture and the disk image: https://steeph.de/stuff/sachen/dinge/zeug/qd/

keirf, 2019-11-24:
There seems to be more than two writes on the final QD image. Is the trace truncated, or was the image not pristine when you started the test?

keirf, 2019-11-24:
I'm not certain the writes are going to the right place, you see, as they end up in the 'non read window' section of the image file. But working out which writes come from where is difficult when there are more writes in the image than indicated by the Saleae trace.

keirf, 2019-11-24:
A useful test setup:
1. Start with pristine QD image (not previously written at all)
2. Run the logfile firmware (under alt/quickdisk/logfile/)
3. Perform test, gather Saleae trace
4. Eject disk image (eg. press Gotek buttons) then pull the USB stick and save FFLOG.TXT

Now we have a disk image, FlashFloppy log file, and Saleae trace.

steeph-k, 2019-11-24:
I didn't realise that I used the image more than twice. This is only initialisation to a copy of Blank.qd:
https://steeph.de/stuff/sachen/dinge/zeug/qd/init.qd

And this should be a document named "blub" containing only the "test".
https://steeph.de/stuff/sachen/dinge/zeug/qd/write.qd

steeph-k, 2019-11-24:
> And this should be a document named "blub" containing only the "test".
> https://steeph.de/stuff/sachen/dinge/zeug/qd/write.qd
I think it's weird that it contains something at all since I see nothing on WD during the entire attempt to save the document.

> A useful test setup:
OK, I'll do this. But first I'll be away for an hour or so.

keirf, 2019-11-24:
> I think it's weird that it contains something at all since I see nothing on WD during the entire attempt to save the document.

Notice that some successive reads after the full initialisation write also themselves contain small "writes".

keirf, 2019-11-24:
init.qd does indeed contain only one write. It looks totally fine.

write.qd is entirely blank -- contains no writes at all??

steeph-k, 2019-11-24:
> write.qd is entirely blank -- contains no writes at all??

You're right. So nothing weird about it.

So, I've done some tests. But initialising is the only thing that will write anything to the image. Everything else looks the same (five attempts to read something), whether the disk was initialised before or not.

So the only interesting test to look at is probably a:
https://steeph.de/stuff/sachen/dinge/zeug/qd/a.qd
https://steeph.de/stuff/sachen/dinge/zeug/qd/a.log.TXT
https://steeph.de/stuff/sachen/dinge/zeug/qd/a.logicdata

I've also done one where I tried initialising again after it failed once. Nothing different in the second attempt. But maybe the resulting image is somehow interesting, idk.
https://steeph.de/stuff/sachen/dinge/zeug/qd/i.qd
https://steeph.de/stuff/sachen/dinge/zeug/qd/i.log.TXT
https://steeph.de/stuff/sachen/dinge/zeug/qd/i-1.logicdata
https://steeph.de/stuff/sachen/dinge/zeug/qd/i-2.logicdata

> EDIT: I do think we're closing in on this. :)

That sounds great! I'm really grateful that you're looking into it. I mean, apart from the work that already went into it.

keirf, 2019-11-24:
The real weirdnesses in the earlier image are gone. I can't really identify anything very obviously bad now. There are two writes by the way: Look at the write gate getting asserted at the end of the first "read attempt" in a.logicdata.

Perhaps it is worth perturbing the initial image a bit and see what happens. Attached here is a QD image with a 5.0-second "read window" rather than the default 5.5s:

[5.0.zip](https://github.com/keirf/FlashFloppy/files/3883805/5.0.zip)

We could also try a slightly longer one:

[6.0.zip](https://github.com/keirf/FlashFloppy/files/3883807/6.0.zip)

steeph-k, 2019-11-24:
I'm getting more variation with 6.0. Collected some recordings but didn't save them before Logic crashed. So here are just two:

6.0-2 was the longest I've got in about 10 - 15 tests. (The Saleae recording is cut off because I didn't expect it and had the maximum recording time set too low.)

I can't tell from the difference of the three blank images that I have how I could try larger times. 6.0.qd seems to have a larger number than 5.0.qd but then the original Blank.qd has an even larger one. So I'll just wait for more images from you.

keirf, 2019-11-25:
If you can run python and clone the FlashFloppy github repo, you will find a script: scripts/mk_hfe.py. It takes parameters to configure the "read window" and "total time". So you could play with that.

By the way there were no attachments on your latest post. Sad. :(

steeph-k, 2019-11-25:
Oh, looks like I forgot to add this link: https://steeph.de/stuff/sachen/dinge/zeug/qd/6.0/

I'll try the script when I'm back from work.

keirf, 2019-11-26:
It does seem to make a difference. Perhaps let's try a mnuch bigger window. Like:
```
python scripts/mk_qd.py --window=11.5 --total=14.0 11.5.qd
```

steeph-k, 2019-11-26:
There is no mk_qd.py and mk_hfe.py takes different arguments.

steeph-k, 2019-11-26:
I mean: I'll do some more testing later. But I can save and load documents with this image. That's all I need. Thank you very much for your great help!

Do you want logs, images and recordings from the logic analyser? I think I'll try longer and shorter read windows too. But 11.5 at least works.

keirf, 2019-11-26:
No need for logs. Please do just try varying the window... I would like to then add a new blank image to the next release and some more instructions to the wiki.

steeph-k, 2019-11-26:
Actually, today everything from 6.0 up works, including the image you provided earlier that didn't work before. So maybe it would be saver to use 6.5 or more. I've only tried .5 steps so far. I can't see any difference except for the time it takes to read.

keirf, 2019-11-26:
And 5.5 (the original default) never works?

steeph-k, 2019-11-26:
It didn't work today. Maybe next week? :)

keirf, 2019-11-26:
Maybe give window 6.5 total 8.0 a good kick around and I'll ask Roland guys to test that too. Would be nice to have a single default working blank image for everywhere.

keirf, 2019-11-27:
Here is a Blank 6.5-second window image:

[Blank_6.5.zip](https://github.com/keirf/FlashFloppy/files/3898353/Blank_6.5.zip)

keirf, 2019-11-27:
^ This image has been successfully tested on Roland MT-100.

@steeph-k Can you give this specific image a test on the Triumph Adler? It is candidate for inclusion in next FlashFloppy release.

steeph-k, 2019-11-29:
My work hours are weird, sorry.

I'll test it more thoroughly over the weekend. So far no problem.

keirf, 2019-11-29:
No rush, just keeping the bug tickets ticking along.

steeph-k, 2019-12-01:
I was somehow able to create a "damaged file". But I don't know how and I didn't see anything on the logic analyser. I can't reproduce it either. I think I've just pressed buttons in the wrong row or too quickly after another. I need to get a manual for this thing. Other than that it's working nicely.

keirf, 2019-12-02:
That's great. This new default image will be in release v3.8a. Thanks!

closed 2019-12-02 by keirf