keirf/flashfloppy #283: Images just cannot boot with the Gotek

slackwaree, 2019-11-13:
Hello,

I have this strange issue with my Gotek that the images I put on are not booting.
For example I have tried over 100 .imgs from winword like:

https://winworldpc.com/product/windows-98/98-second-edition

These images boot with VirtualBox or Vmware just fine, with the Gotek not a chance.

First I had the factory software on it and I used that USB Flash drive manager from win10 to write out the images, didn't work so I switched to Flashfloppy latest stable 2.x, that didn't work either so I switched with the latest 3.x dev version it's the same.

What I have tried:
-2 different motherboards
-different USB sticks

Unfortunately the boards I had didn't have tons of options for floppy boot except boot order, one was an old Aopen the other an old Gigabyte board. 

Once I boot an OS on those machines like Windows 98 or Linux they can read/write the images fine and I can switch images but whatever I do the Gotek just does not boot.

For example from Windows I did format A: /u /s when it formats the virtual image fine and puts the basic boot files on it but no luck. I have a feeling this is not a hardware problem and there is some small software issue going on here.

I have also tried without FF.cfg. With FF.cfg leaving things on default:
interface = jc
host = unspecified

and tried:
interface=ibmpc
host=pc-dos

Any tips are welcome.

keirf, 2019-11-13:
If you can access the floppy okay after boot then the problem cannot really be on the Gotek side. In case the Gotek is not ready early enough after power on (unlikely) have you tried soft resetting without power cycling, and see if the machine boots from floppy then?

Are you sure there are no hidden floppy / integrated device configuration options in the BIOS? Sometimes these things can be disabled or in the wrong mode until properly set up by the OS drivers.

Have you tried booting from a real floppy drive (if you have one!)?

tenuke, 2019-11-14:
What you linked as an example are not floppy images but some 500 megabytes
images. I dont think that those ought to work with gotek.
So is the problem there with normal images of a floppy disks too?

slackwaree, 2019-11-20:
Ok sorry for the wrong link, the thing is whatever boot floppies I tried didn't work so now I made a clear test with the following images:

Win98 SE
https://winworldpc.com/product/microsoft-windows-boot-disk/98-se

MSDOS 6.22
https://winworldpc.com/product/ms-dos/622

Windows XP
https://www.allbootdisks.com/download/xppro.html

Windows NT 4.0 Workstaion
https://www.allbootdisks.com/download/nt.html

Hashes:
4aaa24e3c383a3d27a9a0009ee65a38d  Windows 98 Second Edition Boot.img
64ccadff9eea0feac8ac4947fbe4dbf9  WindowsNT_4.0_Workstation_Disk1.img
6979305571ab65c4b0fd2f9410ee84df  WindowsXP_Pro_Disk1.img
bc929d88acd0bbbd6734cd18e1a633f2  msdos622_disk1.img

Each and every of these I have verified to boot fine with Vmware Workstation 15.
Here is what I get at boot:

Dos/NT

Non-System disk or disk error

Win98

Invalid system disk

XP
IO error reading disk

One thing I did find in the bios is: Floppy 3 mode > DriveA/Disabled. I have performed the test in both settings, does not make any difference.

After booting Ubuntu on the hosts ALL the floppy images including the Windows XP one are perfectly readable and writable. I start to get fed up with this, I have spent a lot of time modding my Gotek both software and hardware vise (I did all mods like LCD, speaker, potmeter).

keirf, 2019-11-20:
Have you tried booting from soft reset? Have you tried a real floppy drive?

NitroXpress, 2019-11-21:
Which Gotek model do you have ?
SFR1M44-U100 ?

I have 3 of them, and they all work like a charm on different systems (2x PC and 1xRoland W30 Sampling workstation)

Many Thanks to keirf and this great project !

Frank

slackwaree, 2019-11-21:
Soft reset yes because someone had an issue with Gotek on vogons where the Gotek "warm up" too slow for that old machine and it did not boot. Real floppy? give me a break drives I have floppies 0.

SFRC992D

keirf, 2019-11-21:
How far do the different images you tried get through booting? I noticed you get a different error message from each one, so it must have read the boot block and be running code off the respective disks... But then fails. It is odd.

NitroXpress, 2019-11-21:
SFRC922D was intended for use with Commodore Amiga.
Jumper plugged on S1 ? (Amiga use S0)

slackwaree, 2019-11-22:
Ok I guess there we have the answer. I didn't even bother checking versions since I assumed all Goteks are the same when I bought mine. Now that you said it I went back checking my purchase history and found: 

SFR1M44-U100 3.5" 34pin Floppy Disk Drive USB Emulator Simulation For Music Keyboard DY

So if I buy another one which should I look for? 

Yeah it was on S1 jumper, if I put it on S0 then it doesn't even get that far (no floppy seeking led/noise), the drive is completely ignored at boot.

drdpj, 2019-11-22:
If you stuck flashfloppy on that model of gotek it should work.

That it isn't behaving at all does suggest that the gotek itself might be
non-functional. Or the PC. Models to buy are listed in the wiki, but the
one you previously bought is OK.

NitroXpress, 2019-11-22:
drdpj is right...

I bought my Gotek as SFR1M44-U100.
Now I see on the PCB SFRC922D...
So it should be the same as yours !

I only have Jumper on S1.
JC configurated over ff.cfg as ibmpc
So your Gotek should actual work...

Volutar, 2019-11-23:
Aren't there HFE images for that purpose? They would indicate if it's either HW/jumper or image problem.

keirf, 2019-12-02:
Not clear there's anything more to say here. Closing.

slackwaree, 2019-12-15:
Although you have closed this NitroXpress was right.

His was the right answer, it was not because of the cable or motherboard or bad usb sticks. Just a simple configuration issue.

FF.CFG had to be in the FF directory and emulation type set to ibmpc.

After this all 4 boot disks worked like a charm. 

Maybe you should change the next version to search FF.cfg on both places also default to DOS format if it doesn't exist since I belive 90% of the people who using this emu are using it for retro DOS machines.

NitroXpress, 2019-12-15:
Great !
I'm happy for You...

Meanwhile I bought two more Goteks and they all work like a charm. (Commodore Amiga 500 and Atari 520 STFM)

Frank

keirf, 2019-12-15:
Amiga/ST/CPC dominates massively. All Shugart.

It is indeed ambiguous that if you have a FF/ subfolder, config Giles are only searched for there. I will update the wiki and the example config files to make this clear.

keirf, 2019-12-27:
> It is indeed ambiguous that if you have a FF/ subfolder, config files are only searched for there. I will update the wiki and the example config files to make this clear.

Done (in the wiki, and in master for next v3.x release)

closed 2019-12-27 by keirf