keirf/flashfloppy #286: SCI Prophet 2000 compatibility

baritonomarchetto, 2019-11-25:
Hi, I am trying to have a gotek drive with FF firmware installed working with a Porphet 2000 synth. The synth was equipped with an EPSON SMD 300 floppy drive (DS/DD disks) which is one of those working on amiga ST, if I remember well.
I have correctly installed the latest (3.7a) firmware, and FAT32 formatted a (old, 1Gb) pen drive.
When I turn the synth on, I get a E-34 error.
If I try to save a new file from the synth itself, it returns "nd" which means "no disk".
Is this a pure incompatibility issue or is there a way to workaround this?
Thanks

keirf, 2019-11-25:
FlashFloppy works with disk images, so you need image files on your USB stick. Do you know anything about the disk format used by Prophet 2000? I think it's quite proprietary and you will need to use an HFE image file. I attach a blank one here. You load this onto your USB stick and it is automatically mounted at power on. You will then need to **Format** it on the Prophet 2000 before you can save to it.

Apart from that, make sure you have a jumper at the rear of the Gotek at **S0** only.

[prophet.zip](https://github.com/keirf/FlashFloppy/files/3886162/prophet.zip)

baritonomarchetto, 2019-11-25:
Yes, I have a jumper at S0 and no other jumpers.
Even with the image unzipped in the pendrive root folder, I still have a E34 error

keirf, 2019-11-25:
So you have a .hfe file now in the root of your USB drive? Not in a subfolder? It should be impossible to get E34 in this case.

baritonomarchetto, 2019-11-25:
Yes, I have the hfe file you linked in the root folder and still E34 error. I will try to copy it in a "FF" folder and see what happens. I will try another pendrive too.

(Thanks for the help so far, much appreciated!)

keirf, 2019-11-25:
Yes very odd: If you have one single file on your USB stick, called prophet.hfe, then it should definitely mount correctly. Just tried that very image myself.

baritonomarchetto, 2019-11-25:
just tryed another pendrive (4Gb): no luck. I am going to install an older flash floppy version, but I am starting to think this floppy could be damaged... :(

baritonomarchetto, 2019-11-25:
rolling back to ver 2.13 actually worked! The smaller pendrive (1Gb) is still not recognized, but the 4Gb one works like a charm. Thanks!
Now, am I correct assuming that placing new, renamed "prophet.hfe" files the number of slots (floppy) will increase accordingly?
Great work you did man...

keirf, 2019-11-25:
That is correct. Which version was not working for you by the way? I will double check that version myself.

The firmware can clone an image, but you require the "Eject Menu" functionality for this which requires v3.x firmware and an OLED screen (recommended anyway if using quite a few images, unless you run in Indexed mode and keep manual notes of which index numbers map to which disks).

baritonomarchetto, 2019-11-25:
I was using the 3.7a version.
I have seen the OLED screen mod, but I like the standard LED segments display more. Consider that the synth itself mounts red led, 3x7 segments display.
Could a basic version of this feature something implementable for standard display? It would be sufficient a clone file up to the USB pendrive limit capacity..
(thanks a lot!)

keirf, 2019-11-25:
I suggest you configure Indexed mode (https://github.com/keirf/FlashFloppy/wiki/Initial-Setup), and create a bunch of DSKA0000.HFE, DSKA0001.HFE, etc images on the stick manually on your PC. Then keep track of what you have written to each number :)

keirf, 2019-11-25:
Okay I think this ticket is done.

closed 2019-11-25 by keirf

baritonomarchetto, 2019-11-25:
Yes, it's done. I just wanted to say that I have upgraded to the latest firmware and the device still works, so it's likely I had a pendrive error in first instance.
Thanks for this great piece of software and help!