keirf/flashfloppy #287: Better sound ? Use the DAC ?

penfold42, 2019-11-25:
Is there enough cpu time spare to use the DAC ?

A4 and A5 are unused on the gotek.

I miss the swish swish swish click swish swish swish click

Or another option - use a F105 for the OSD and get it to do the hard (or not so with DMA) work

Volutar, 2019-11-25:
You mean rustling sounds of rotating disk?
I doubt there's a CPU time for that.
But I think it is possible to add some kind of shortest clicks at random occurances.
Though I doubt this sound will be pleasing.

keirf, 2019-11-25:
Maybe one day :) Especially if ported to a bigger board.

bradwh, 2019-12-17:
Have you thought of using something like DFPlayer Mini with a sampled sound file. Then use the JB signal to trigger the playing of a short sampled sound from a real FDD. Ideally the sampled sound could be a head step sound followed by the disk spin sound "rustling sound of rotating disk as Volutar calls it, which lasts for about 2 seconds then goes silent as would a normal floppy after access. Each time the head step signal is output from JB the sound track is started from the beginning again.