keirf/flashfloppy #288: D88 format support

tdaede, 2019-11-25:
This is a format used for various Japanese computer images (including the PC-98).
https://fjkraan.home.xs4all.nl/comp/px4/vfloppy/d88Format.html
Unlike other PC-98 formats, this one supports varying sector size and density on a per track basis, so it will be a bit more complex to implement.

SuperTurboZ, 2019-11-26:
Many Japanese users will want to support D88.
Since D88 needs to move a lot of track data depending on the track write, it is not suitable for emulators with small RAM.
I think it is better to support read-only or automatically  pre-convert to HFE .

tdaede, 2019-11-26:
My plan is to support reading and writing, but not formatting. All of the structures are fixed length as long as you write with the same sector size as it was formatted, so it should work out OK.

Peredonov, 2024-08-12:
I can see it's been a while since D88 support was requested. While waiting for it, is conversion to HFE recommended, or is there a better format to convert D88 files for use with Flashfloppy?

Feinsveur, 2024-12-27:
Hopefully D88 support is added. Currently most software works well converted to HFE on a dual Gotek setup, however a few won't boot or have some issues.