keirf/flashfloppy #289: Wiki Update for Host Platforms: Acorn Archimedes A3020

dandoore, 2019-11-26:
I have found that the Archimedes A3020 (and presumably any of the HD floppy capable models post A5000) uses an IBM drive rather than a Shugart type - tip off from https://torlus.com/floppy/forum/viewtopic.php?f=19&t=562&start=30#p16503

All the other settings are fine, but this tripped me up so can the Wiki be updated to show this :)

keirf, 2019-12-02:
Now updated to mention the need for `interface = ibmpc`. Is there anything else to mention, such as different drive-select jumper setting?

Please reopen and let me know what to add, if necessary. Thanks!

dandoore, 2019-12-02:
Just checked, I had to move jumper to S1, so it's the same for the generic IBM-PC interface bit: 

- IBM-PC interface mode must be configured
- Strap jumper JC at the rear of the Gotek; or
- Specify via FF.CFG: interface = ibmpc
- Strap select-line jumper S1 at the rear of the Gotek
- S0, S2, MO should all be left open

dandoore, 2019-12-02:
Bugger, I can't reopen this issue :(

keirf, 2019-12-10:
Fixed, thanks.

closed 2019-12-10 by keirf