keirf/flashfloppy #291: VFO read window for certain 8" floppy controllers

jboone, 2019-12-01:
Hi! This is a fantastic project and I'm very grateful for all your work on it.

I have an [NEC APC computer](http://www.oldcomputers.net/nec-apc.html) that is a PC-DOS-ish machine from the early 80s. It has DSDD 8" drives, and uses the [8" IBM FM and MFM formats](https://en.wikipedia.org/wiki/List_of_floppy_disk_formats). The trouble is, the *drive electronics* contain a VFO that provides a "read data window" (synchronized to the read data) over the ribbon cable, to the floppy controller. I believe other systems using the uPD765 controller (including the IBM PC) chose to integrate the VFO into the controller, not the drive. Hence, this VFO and read window business is not a concern on most systems.

Here's an [application note for the uPD765](http://www.bitsavers.org/components/nec/_dataSheets/uPD765_App_Note_Mar79.pdf) that describes the data window requirement, starting on PDF page 17 (document page 16).

Essentially, the read window is a square wave that toggles at the symbol rate of the read data stream. The square wave polarity is apparently not of concern, but the wave transitions should be at the boundaries of the bit cell period. The flux transition pulse should be squarely between the read window edges.

I reviewed the FlashFloppy code, and think I could appropriate a timer and I/O pin to do this. Initially, I was thinking I could use another compare register on TIM3, but the period of TIM3 (ARR value) is changed depending on the symbol coming from the floppy emulator -- whether the bit cell has a flux transition or not. Perhaps I can start another free-running timer synchronously with TIM3. This assumes that the bit cells are all the same period, which is I believe is not guaranteed for flux formats like HFE. I suspect I will need to re-synchronize at the start of every sector.

Any thoughts on this? I do a lot of open-source ARM embedded code and am fairly familiar with the STM32F product line, so could probably make this work, one way or another, with some significant head-bashing. Of course, any wisdom before I set out on this journey would be greatly appreciated!

(The [NEC APC System Reference](http://bitsavers.org/pdf/nec/APC/819-000100-1003_APC_System_Reference_Guide_Apr83.pdf) might be of interest. The floppy system is described on PDF page 86 / document page 3-21, and the floppy controller schematic is on PDF page 230 / document page B-22.)

keirf, 2019-12-01:
Do you actually need this extra signal? We can do the data separation on
the host exactly as we do for any other drive. The serial stream off disk
is still self clocking. Grab a dump and try sticking it through hxc
software. Bear in mind you probably need to buffer greaseweazle outputs to
talk to an eight inch drive

keirf, 2019-12-01:
That's an interesting one. Actually HFE is uniform bit rate. FF individual bitcells may vary +- a tick because we track fractional ticks, but ignore/disable that and you could probably get away with syncing the two signals at the index and/or hard sector marks. You do need to start the timers at same instant. Possibly doing do manually with irqs disabled is good enough, else perhaps you could get one to trigger the other.

jboone, 2019-12-01:
@keirf Thanks for the response!

It's good to hear HFEs might not be as problematic as I'd assumed. I need to learn more about HFE and IMD and IMG. I have other problems around the RPM and bitrate of the NEC APC IMDs I'm converting, and understanding the formats might clear up that issue for me.

The ST reference manual for the F10[57] series has some good notes on synchronizing multiple timers. There looks to be a way to have one start the other via a hardware/peripheral-to-peripheral connection.

I'll give it a try today and see what I can do. Thanks again!

keirf, 2019-12-02:
Yes you may be able to get RDATA to start, kicked off by the WINDOW timer. Have fun ;)

jboone, 2019-12-02:
I got it to work. I used a hardware mechanism to enable TIM2 from TIM3's TIM_CR1_CEN. They're about 15ns out of phase, which is good enough for me. I preset the TIM2 counter to skew the timer output to be at the edge of the cell time. After some unrelated fiddling, and being caught by some issues around HFE creation and implementation (bitrate and RPM, particularly), I got the NEC APC to boot MS-DOS 2.11 just fine.

I'll fork and clean up the code and commit so you can have a look and see what's useful. Given that I had to appropriate the READYn pin for testing, and TIM2 channels aren't available elsewhere on a GoTek, this patch may be just for me, until someone (me?) comes along with some enhanced hardware to support it.

Once I've pushed, I'll drop a link to the commit here and then close the issue

jboone, 2019-12-02:
I've pushed my NEC APC read window changes to https://github.com/jboone/FlashFloppy/tree/nec_apc_rwin

I also have a small fix for the ARM GCC 9 I'm using. There's a new warning regarding the strncmp size_t argument being too large (~0). I can send you a PR if you're interested. https://github.com/jboone/FlashFloppy/tree/gcc9_strncmp_warning

keirf, 2019-12-02:
Leave this open for now and I'll take a look. Retasking pin 34 is not necessarily a problem as pins 2 and 34 are both configurable.

The gcc9 fix I think I already have in my Greaseweazle project and should port across.

keirf, 2019-12-10:
The gcc9 fix is now in FlashFloppy. Please check it works okay for your environment.

keirf, 2019-12-10:
Your patch is pretty tidy. It just needs to be hidden behind a configuration option. The neatest would be to be able to say `pin34 = window` in FF.CFG.

You could define a new PIN_window in inc/config.h. It doesn't need a corresponding outp_window because this is rather a special pin assignment behind the curtain: It's not really a normal 'outp'.

So I would specifically check for ff_cfg.pin34==PIN_window in floppy_set_fintf_mode. I would then set pin34 to outp_unused and take direct control of that pin with your new code. You could then gate your code elsewhere on ff_cfg.pin34==PIN_window.

ff_cfg.pin02==PIN_window would be invalid. I guess set pin02 unused in that case.

Does that make sense?

Tachiorz, 2019-12-12:
Some NEC PC-98 models has this problem as well, they won't work without window signal present. I implemented simple external circuit based on counter mentioned in [application note for the uPD765](http://www.bitsavers.org/components/nec/_dataSheets/uPD765_App_Note_Mar79.pdf) and it works well for 500kbps HD images with 8mhz crystal. But to support DD images I'll have to add another crystal and to support FM mode I'll have to divide frequency and it's becoming a mess.
I would like FlashFloppy to support window signal natively for variety of bitrates.

Tachiorz, 2019-12-13:
Also, is it possible to make it work on pin02? READY might be used by some PC-98 games.

keirf, 2019-12-13:
No it is tied to pin 34 but you could assign READY to pin 2 and then physically remap that pin via modified ribbon cable or whatever.

Tachiorz, 2019-12-14:
Flashed nec_apc_rwin and can't figure out how to make it work. I tried `pin02=rdy pin34=nc` and pin34 doesn't change at all. with `pin34=low` it doesn't look like a window.

keirf, 2019-12-14:
The patch doesn't really respect the drive select line very well. Does it work if you permanently select the gotek (tie the select line to 0v)? I suspect the author tested the patch in a permanently-selected setup

Tachiorz, 2019-12-14:
I tried setting MO jumper and it does makes window signal now, but it doesn't look centered. https://i.imgur.com/BvUsr1J.jpg

keirf, 2019-12-14:
It does not look a million miles off? It's the falling edge of RDATA that matters, and it's not super close to window edges? Does it work?

Tachiorz, 2019-12-14:
It doesn't work. Shouldn't be very precise, this one is captured from gotek with external circuit and it works fine. https://i.imgur.com/Mc0t72F.jpg

keirf, 2019-12-14:
Those low pulses on RDATA are much shorter in the second trace aren't they? Different firmware?

Tachiorz, 2019-12-14:
That was stock FlashFloppy firmware, RDATA was synced with external circuit. See uPD765 App Note (from this thread) page 23.

Tachiorz, 2019-12-14:
And this is an example from native drive https://i.imgur.com/9UWr9jE.jpg

keirf, 2019-12-14:
I would try adjusting CNT register in the first instance, as that is used to centre the window. Apart from that you can fine tune the rate by initialising ARR and CCR2 in rdata_start based on ticks_per_cell (perhaps that value >>4 and subtract 1).

Tachiorz, 2019-12-14:
This is what I put in rdata_start()
`    tim_rdata->cnt = 0;
    tim_rwin->arr = (drive.image->ticks_per_cell>>4) - 1;
    tim_rwin->ccr2 = tim_rwin->arr;
    tim_rwin->cnt = sysclk_ns(320); /* Skew to center RWIN on bit cell time. */
`
It goes out of sync pretty fast https://i.imgur.com/eVWIb6l.jpg

keirf, 2019-12-14:
All that stuff wants to go before the write to EGR. I also recommend `ticks_per_cell &= ~15` to make the bitcell timing a proper multiple of sysclk.

Tachiorz, 2019-12-14:
```
    drive.image->ticks_per_cell &= ~15;
    tim_rdata->cnt = 0;
    tim_rwin->arr = (drive.image->ticks_per_cell>>4) - 1;
    tim_rwin->ccr2 = tim_rwin->arr;

    /* Start timers. */
    tim_rdata->egr = TIM_EGR_UG;
    tim_rwin->egr = TIM_EGR_UG;

    tim_rwin->cnt = sysclk_ns(360); /* Skew to center RWIN on bit cell time. */
```
Much more stable, but still not in sync. Manages to read few sectors slowly, eventually fails to boot.

keirf, 2019-12-14:
Does it still drift?

Tachiorz, 2019-12-14:
https://i.imgur.com/sR8MCGJ.jpg

keirf, 2019-12-14:
That's a no. Then you need to futz with CNT initial value.

Tachiorz, 2019-12-14:
Isn't it just an offset to center window? It was okay to the end of track 0 https://i.imgur.com/Bcb1e60.jpg, and a bit later it's skewed https://i.imgur.com/UZt7uvD.jpg

keirf, 2019-12-14:
If its not drifting but staying constant offset from where it is at rdata_start then CNT is currently the best you have to work with. If its varying track to track then perhaps something smarter is needed. If its drifting within a track then ARR isn't quite right.

Tachiorz, 2019-12-14:
I tried ARR -2,-1,+0,+1. -1 seems to be closest. But I think it's varying from track to track

keirf, 2019-12-27:
Sorry for delay. Perhaps try the `&=~15` adjustment of ticks_per_cell where it gets initialised, in hfe.c. Perhaps it is getting "fixed up" too late.

keirf, 2020-01-17:
This seems dormant. Reopen if there is interest in merging this into mainline.

closed 2020-01-17 by keirf

dumuge, 2021-08-21:
Excuse me, can the current version realize the window of vfo with 34 pin?

keirf, 2021-08-21:
> Excuse me, can the current version realize the window of vfo with 34 pin?

No this feature is not in a FF release currently.

dumuge, 2021-08-21:
> > 请问现在的版本能实现34pin的vfo的窗口吗？
> 
> 不，此功能目前不在 FF 版本中。

> > Excuse me, can the current version realize the window of vfo with 34 pin?
> 
> No this feature is not in a FF release currently.

So, which release can I try？

dumuge, 2021-08-21:
> > Excuse me, can the current version realize the window of vfo with 34 pin?
> 
> No this feature is not in a FF release currently.

So,  Which release can I try?

keirf, 2021-08-21:
> > > Excuse me, can the current version realize the window of vfo with 34 pin?
> > 
> > 
> > No this feature is not in a FF release currently.
> 
> So, Which release can I try?

You have to build it from @jboone forked repository linked to previously.