keirf/flashfloppy #292: Eject Menu Clone Image

DOSputin, 2019-12-02:
Been trying to clone blank `.HFE` images on an **E-mu Emulator II**.
I have a usb disk prepared, and a blank image named blank.hfe
From the eject menu, I select the `blank.hfe` image, navigate to `Clone Image` and select `Confirm Clone`.
At this point nothing appears to happen, and I am returned to the `Clone Image` menu entry.  
Inspecting the disk, there is no clone.  

Is there anything that I am missing?

keirf, 2019-12-02:
You should find your clone named blank_000.hfe iirc

DOSputin, 2019-12-02:
That is what I expected to find.  But it seems the the clone isn't ever created.  
As I said:
>Inspecting the disk, there is no clone.

keirf, 2019-12-02:
Latest v3.7a firmware?

keirf, 2019-12-02:
You should get a message on the OLED display for at least two seconds. It should say `Cloning to 000`.

If you get no message then the `Confirm Clone` may have failed. This should succeed if you pressed the rotary-encoder/third/select button. Do you have that button? If you are resorting to pressing the two basic buttons together, that is trickier timing, but then again you managed to navigate the Eject Menu okay...

DOSputin, 2019-12-03:
>Latest v3.7a firmware?
Yes.  The drive in question was on 3.7a

>This should succeed if you pressed the rotary-encoder/third/select button.
I was attempting this on my EII that does not have encoder / 3rd button. 
Attempted with the same USB, in a Oberheim DPX-1 that is equipped with Rotary Enc. /w 3rd button and it worked first try. 

I then went back to the two button drives and gave it a try, and used two fingers to press the buttons to `Confirm Clone` and I got it.  
So it appears that my timing was consistently off on the 2-button drives.  Maybe a programatic button press offset of a few MS would help resolve this?    I'm an old fart, and have arthritis in my hands so I'll blame that for now.   The two finger method works for the time being.  

Thanks for the input, I wouldn't have though about trying on a drive with the 3rd button.   
Now I've got more work to do this weekend to take the Eii apart and upgrade it's drives with encoders.  
Joy!

closed 2019-12-03 by DOSputin

keirf, 2019-12-03:
There is a 50ms wait for both buttons to become pressed, after the first is pressed. This is a tradeoff between accepting two-button presses, versus lag on single-button presses.

If you would like a better input method I would recommend a rotary encoder, which includes a push button too.