keirf/flashfloppy #293: Can a special HD interface mode for Amiga 4000 Tower be supported?

tkurbad, 2019-12-05:
Hi,

I recently built an Amiga 4000 Tower mainboard out of a blank new old stock PCB.

The nice thing (and at the same time problem) with the A4000T is that it's modular, i.e. most of the I/O ports are placed on pluggable module PCBs. This is true also for the disk module that holds the connectors for SCSI and floppies.

The original modules, however, are hard to obtain (= the aforementioned problem), so some creative guys rebuilt the disk module with a more modern approach. It is outlined [here](http://www.amigawiki.de/doku.php?id=de:projects:a4000t_diskmodul) (German only, unfortunately). To summarize, the rebuilt module includes a CPLD that makes it possible to use unmodified PC drives in DD mode on the Amiga just by setting a jumper. The RDY signal needed by the Amiga is in this case created by a microcontroller on the disk module itself.

The most interesting feature is the HD mode. To use it, you have to modify a Sony MPF 920 floppy drive to:

1. output the state of the HD switch to pin 2 of the 34 pin header
2. output the clock/rpm(?) signal of the motor to pin 4
3. feed the signal from 2. back via pin 6 into the drives circuitry (I assume)

The reason for 2. and 3. is that the Amiga can read HD floppies at half the normal speed only. Thus, circuitry on the disk module divides the motor clock signal and feeds it back into the drive for the motor to actually turn at half its normal spin rate. (At least that's how I understand the explanation)

The downside to this hack is that all connected drives have to have the same properties, i.e. all must be of the same type (Amiga or PC) and all must support the same density (DD / HD).

What I'm now trying to accomplish is to use a modified Sony drive and a Gotek at the same time. But the setup already fails with the Gotek alone.

I tried setting the Gotek's interface to `ibmpc-hdout` but then the drive never stops loading, even for DD *.adf images. I guess this is because of the missing signal on Pin 4.

In consequence, I've got a couple of questions:

1. Does the hdout signal on pin2 correctly differentiate between DD (880 kB) and HD (1760 kB) amiga disk files in this interface mode?
2. Would it be possible to send the data with half the normal rate when HD mode is asserted? Could an appropriate motor pulse be generated on pin 4 of the floppy interface? Is this pin even connected on the Gotek? Is this the same signal as the one that can be routed to pin 16?
3. Do I possibly need the MTR signal hardware mod anyway in this configuration?

Many thanks in advance!
Torsten

tkurbad, 2019-12-09:
I've had partial success now. The Gotek detects and reads Amiga DD floppies in `ibmpc` interface mode while the `HDFix` jumper on the disk module is in place. Previously, this configuration only yielded read errors.

To accomplish this, I first did the Pin 16 motor hack, although I don't think it's needed.

The relevant sections in `FF.CFG` look as follows:
```
interface = ibmpc
host = unspecified
pin02 = ndens
pin34 = chg
motor-delay = 200
```

In this configuration, all images are detected as DD, which obviously works for real DD images but not for HD.

In theory, this config should work, but it does the exact same thing as above:
```
interface = amiga
host = unspecified
pin02 = ndens
pin34 = chg
motor-delay = 200
```
I'd choose `interface = amiga` because this already has the HD specialties for Amiga implemented.
My guess is now that in HD mode, the disk module expects the 1 MHz motor CLK signal on pin 4 (which is then divided by 2 and fed back into pin 6). In particular, the VHDL code looks for a rising edge on pin 4 that obviously never occurs.

I guess what I have to do to make this work is to add an 1 MHz oscillator and feed the output into pin 4. I'll try that and report back...

keirf, 2019-12-09:
Looking at FlashFloppy source code, src/floppy.c:303, the ndens output is toggled based on a bitcell timing threshold of 1.5us. This is halfway between the normal HD and DD timings of 1us and 2us respectively.

But, HD Amiga drives always have a timing of 2us (because they spin HD half rate). So this won't get triggered and the ndens output will always be constant.

By the way are you sure you want ndens rather than dens? Confusing, but dens is already active LOW (ie. outputs LOW/0v for a high-density image). So ndens outputs LOW/0v for a DD image.

tkurbad, 2019-12-09:
> By the way are you sure you want ndens rather than dens? Confusing, but dens is already active LOW (ie. outputs LOW/0v for a high-density image). So ndens outputs LOW/0v for a DD image.

I just don't think the hack done on the disk module does anything the 'standard' way. On the Sony MPF 920 floppies they use for the original hack, the HD switch output is pulled up to 5V while the switch is open, i.e. HD floppy inserted. The switch closes to GND, thus - in my understanding - connecting the switch directly to pin 2 does behave the `ndens` way, or am I mistaken here?

I tried feeding the CPLD with a constant 1 MHz from pin 4, too. But that doesn't help. The Gotek just reads track 3 and then 2 from a HD image and then gets stuck. I still don't know if this is because of the missing CLK signal on pin 4 or because of a wrong assumption regarding density.

Using the Sony floppy as a reference, 1 MHz on pin 4 is only present while the motor is spinning (obviously) and correctly fed back to pin 6 as a 500 kHz signal while a HD disk is inserted. That should make the disk motor turn at half it's normal speed.

tkurbad, 2019-12-09:
Digging further, I think the `interface = ` config parameter should stay at `ibmpc`.

I now tested this interface setting in conjunction with a HD image and `pin02 = high` and it works!
I successfully read/wrote/formatted a HD .adf image.

So, all boils down to implement a way to reliably set dens/ndens in this specific scenario. @keirf can you think of something?
In essence, on the hardware side there would be the clock divider between pin4 and pin6 that could possibly be evaluated by the STM32. But I think there must be a way to do that solely in software.

tkurbad, 2019-12-09:
I tried to code something in the branch a4000t-hdfix of my fork. Will test it tomorrow with my Gotek.

keirf, 2019-12-10:
Please try the following: At the top of update_amiga_id(), after the fintf_mode check, add:
```
drive_change_output(drv, outp_hden, amiga_hd_id);
```

In FF.CFG you will need to set `interface = amiga` and then override pins 2 and 4 as you need: `pin02=xxx` and `pin34=xxx`.

Let me know how that goes!

tkurbad, 2019-12-10:
Hmm, update_amiga_id doesn't get the drv parameter yet. Should I add it to the function definition and all its calls? In which position (before or after amiga_hd_id)?

tkurbad, 2019-12-10:
This https://github.com/keirf/FlashFloppy/compare/master...tkurbad:master compiles fine, but again only works with DD.

tkurbad, 2019-12-10:
Still no luck (please have a look at both commits).

keirf, 2019-12-10:
What have you put in your FF.CFG?

tkurbad, 2019-12-10:
```
interface = amiga
pin02 = ndens
pin34 = chg
```

DD works, so ndens is staying low all the time.

keirf, 2019-12-10:
Okay, the FF.CFG looks good. I think actually the change to update_amiga_id() needed to be a bit more complex. I've built a patch which you can download below. Try that with your new FF.CFG:

[ff_293_1.zip](https://github.com/keirf/FlashFloppy/files/3945425/ff_293_1.zip)

keirf, 2019-12-10:
Huh no. It doesn't work... Argh.

keirf, 2019-12-10:
Nope I was checking pin 4 not pin 2. It definitely does work :) So please confirm that and I will apply the above patch to the repository.

tkurbad, 2019-12-10:
@keirf, works perfectly! Thanks a lot. Just need to document the settings for FF.CFG somewhere safe. Will possibly add a note to the amigawiki page as well...

keirf, 2019-12-10:
Excellent I will commit to master. I'm not sure where to best document it. It probably belongs somewhere alongside the docs for the disk module, and drive modification docs. ie. it's more suitable there than in mainline FlashFloppy docs really? My Amiga docs are lengthy enough as it is!

commit 5cba7a65 (2019-12-10): amiga: Fixes for 'interface = amiga'.
1. Set the HDEN output according to RPM. The bitcell is always 2us on Amiga.
2. Do not futz with pin34 unless it is otherwise unused.

This fixes FlashFloppy for use with an esoteric disk module for A4000T.
Fixes #293.
closed 2019-12-10 by keirf