keirf/flashfloppy #294: Extended Density support?

matrixchoices, 2019-12-06:
I read there is support for the 2.88MB extended density IMG/IMA but I cant seem to get it to with with my Gotek and FlashFloppy 2.13?

keirf, 2019-12-07:
I'm not sure it actually works as I never tested it. I have a testbed project I should probably add ED tests to. At least I can then check that Flashfloppy keeps up at that data rate.

keirf, 2019-12-07:
What is your test by the way? Reading an existing ED image? Writing? Formatting (this is generally toughest)?

matrixchoices, 2019-12-07:
I set the BIOS as a 2.88MB 3.5in floppy as drive A. I made a 2.88MB IMA on winimage. The computer I have it on has jumpers JC and S1 jumpered. I think MO is jumpered as well. It boots and doesnt give me a Floppy Drive Fail error. Now once I get to the prompt I switch to A: drive and if the selector is on the 2.88MB image it kicks back a Drive not ready Abort/Retry/Fail. If i use other images like 7230k or 1.,44MB it works fine. I tried setting up the IMG.CFG with the 2.88MB parameters with no luck. All iv'e done so far is trying to read a premade image at this point. I did try adding data to the image before I tried accessing it with Flash Floppy but still no luck. Hopefully that answers some questions.

matrixchoices, 2019-12-09:
Any other tests i should try?

keirf, 2019-12-09:
You could try another USB drive or two?

I had forgotten that there has been some ED success: see #277 where a user has had read/write success on Sinclair QL with adapter card (but not formatting, which is why that issue ticket is open).

However the QL format is 10 sectors of 2048 bytes each, whereas PC  format is 36 sectors of 512 bytes each. Although less data per track, it's possible the smaller sectors mean FF struggles to keep up.

I will still add 36*512byte testing to the FlashFloppy testbed.

matrixchoices, 2019-12-09:
Sounds good I will try that today

matrixchoices, 2019-12-09:
Still getting General failure reading drive A: wish I had a sinclair to test....

keirf, 2019-12-10:
Okay, I have done some ED testing myself. It does seem to basically work, as all the bug fixing was to my testbed (a separate project).

I note that the track layout is not quite right, and also perhaps the RDATA pulses need to be a bit smaller and sharper at this data rate. So I have made some firmware fixes, which you could test. Just download the attachment below, and do a Firmware Update using the UPD file contained within:

[ff_294_1.zip](https://github.com/keirf/FlashFloppy/files/3944924/ff_294_1.zip)

If you still have problems it may be worth running the alternative logfile firmware, and gathering logs for inspection: https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

matrixchoices, 2019-12-10:
Will do today!

matrixchoices, 2019-12-10:
Just to confirm the ZIP I downloaded from the link above had the 3.7a firmware with the latest release being 11 Novemeber 2019. Still got a General Fault Reading error. Here are the log files for two usb drives I used
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/3945441/FFLOG.TXT)
:
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/3945444/FFLOG.TXT)

keirf, 2019-12-10:
Okay, and those logs were gathered after trying to read those disk images? There are no warnings in there, so the firmware itself doesn't seem to have been struggling with those USB sticks.

How do you actually connect the Gotek to the PC? My understanding is that real ED drives needed a different cable and/or special jumpering?

keirf, 2019-12-10:
Here is an interesting Application Note from Intel: http://www.pix.net/languard/pdfs/29209302.pdf

In particular look at page 9 (pdf page 13) Section "INTERFACE BETWEEN 82077AA/SL AND THE DRIVE". The last paragraph on that page discusses Media Sense pins **from** the drive **to** the host. Could this be the problem?

matrixchoices, 2019-12-10:
Let me do some hardware/BIOS diagnostics.....I think i may know the problem I have another 386 and 486 i am going to test with

matrixchoices, 2019-12-10:
I can confirm 2.88MB floppy support is working now. I realized that the whole thing might be a limitation of the FDC controller chip not having the ability programmed to access 2.88MB floppies even on a motherboard with a BIOS that has a 2.88MB floppy option. I have tested it on an MSI K7T-Turbo. Gotek attached via 34pin ribbon cable (twist end) pin 1 nearest power connector. I'm thinking I should check FDC controller chip capabilities of the other motherboards I was using because I think they may only have programming for 1.44MB and non standard Microsoft varations programmed.

matrixchoices, 2019-12-10:
Ill do some more testing tomorrow with read/write/formats. Ill flash the LOG firmware alt and have log files in case anything else happens

matrixchoices, 2019-12-11:
Sorry off topic...where can I get a Greaseweazel? and possibly a testbed pcb for flashfloppy?

keirf, 2019-12-11:
Have you had success with the latest release (v3.8a)? I'm uncertain whether to commit the changes I made in the ff_294_1.zip firmware. Probably I will skip them if the current releases work on your PC setup.

The FlashFloppy testbed is a bit arcane to use: basically I use it for regression testing of FlashFloppy, and add extra tests to it now and then. My usual setup would be two Goteks, one with the TestBed PCB soldered on, both in bootloader-programming mode, both with a serial-usb adapter. Then I can program both with latest debug builds of FlashFloppy and FF_Testbed respectively, and gather serial logs from both. If this is of interest I'm happy to supply an adapter pcb for cost of postage.

As for Greaseweazle, all my PCBs are currently sold out, but other users have ordered some and they come in lots of 50 so there should be spares floating about. I recommend joining the Greaseweazle FB group, if you use FB. I am making a new design, and may also order more of the original PCB adapter design too.

matrixchoices, 2019-12-11:
Whats the difference with greaseweazel and flash floppy/gotek? havent really read on about greaseweazel until i saw it on the fb group which I will join! and yes I have had luck on one computer setup I have on a K7T-Turbo motherboard its an athlon XP ....i will futhur test read/writes...i did have issues with formatting which i think is expected. It gave me a general fault at the point where it tried saving unfomat data. i tried to format in a DOS 6.22 enviroment but after hitting Fail a couple of times it started formatting at 2.88 but then i noticed it didnt really format. Ill try it from scratch tomorrow and be sure to send the log files along as well as test other motherboards I have

keirf, 2019-12-11:
Greaseweazle connects to a real floppy drive and allows you to read/write at a very low level, to/from your host PC connected via USB. So it is for archiving, and restoring, old floppy disks. Whereas FlashFloppy of course is a floppy-drive emulator. So they're quite different.

Good that you had luck with one PC. Is that with an official FF release? Please do try latest v3.8a. 

Formatting is very unlikely to work. It is worth gathering logs. There is discussion on fixing this in #277

matrixchoices, 2019-12-11:
Can greaseweazel read amiga and other formats? I have a catweasel MK4plus card i sometimes use for reading Amiga disks on a pc. And yes I updated to the 3.8a firmware before doing more testing so all my Goteks have the new firmware. Ill gather some logs maybe they will help

matrixchoices, 2019-12-11:
Im Frank on facebook btw....i dont see the grease weazel group though

keirf, 2019-12-11:
https://www.facebook.com/groups/greaseweazle

Yes it works at flux level so it can read/write any disk format. It is up to host-side tools to process/produce those flux timings though! Currently the tools read/write SCP (Supercard Pro) flux files, which are usable by various other tools inc. my Disk-Utilities and also the HxC host software.

keirf, 2019-12-15:
Any further reports or work to do here? If all is good I'll close the ticket.

closed 2019-12-27 by keirf

cyberluke, 2021-07-08:
fixed in https://github.com/keirf/FlashFloppy/issues/496