keirf/flashfloppy #296: Akai X7000

baritonomarchetto, 2019-12-09:
Hi, I want to report that the quick disk alternate code works flawlessy with the akai X7000. Notice that the pinout is reversed.
Great work!

keirf, 2019-12-09:
No jumper required at JC? If you can read/write with no errors without it then the answer is no :)
Hopefully only Roland requires JC.

baritonomarchetto, 2019-12-09:
Correct: no jumper at JC

keirf, 2019-12-09:
And by reversed do you mean that the colour sequence is reversed?

baritonomarchetto, 2019-12-09:
The cable is a full grey cable: it has no colors at all. By looking at the printed numbers (both in the quick disk drive connector and cable connector) I measured 5V on QD pin 2, not 9. By connecting the gotek reversed (pendrive NOT connected) I broke nothing anyway (yes: I measured the voltage AFTER the first conncetion ... :) )

keirf, 2019-12-10:
Thanks, this is all updated in the Wiki :)

closed 2019-12-10 by keirf