keirf/flashfloppy #299: S-10 QuickDisk

AlbDores, 2019-12-18:
Hi,
I have a Roland S-10 and MKS-100 that I occasionally use, and when i have googled about emulaters i have found your work and be thrill to get other faster way to put them to work. 
I have follow all the steps descibed on your site, but i am missing something beacuse i cant write or read nothing on the pen drive. 
The oled display the content of pen drive but when i save data S-10 presents errors "Illegal disk" or "Error 2". I am missing something? 
Any help very apreciated.

Squirrel61, 2019-12-18:
Did you flash it with the QuickDisk firmware? It is different from the standard 3.8 and can be found in the download archive, in the "alt" folder.

AlbDores, 2019-12-19:
Yes. I have upgraded Gotek to "FF_Gotek-v3.8a.dfu" with DfuDemo then i have copied "FF_Gotek-QuickDisk-v3.8a.upd" to pen drive and then done the upgrade procedure from USB.
I have check the cables and the connections and are acording with the table of connections on this page project.
One thing i have noticed when saving on Gotek, the display presentes initialy at middle screen "10" then "10w" and stops everything. Then after 5 ou so seconds, S-10 display an error.

keirf, 2019-12-19:
Did you place jumpers at JC and MO at rear of the gotek?

AlbDores, 2019-12-19:
Yes, and remove the jumper S0.

keirf, 2019-12-19:
Try the blank QD image from v3.7a (I changed it a bit in v3.8a). Also perhaps take some pics or short video of your Gotek display during write. Also you could install the quickdisk logfile firmware and gather an FFLOG.TXT log file.

keirf, 2019-12-19:
The QD logfile firmware is under `alt/quickdisk/logfile/`

The logfile instructions are here: https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

By the way I have no report of success on S-10 yet, but I do on MKS-100. I would be surprised if it doesn't just work though!

AlbDores, 2020-01-07:
Success on both, Roland MKS-100 and Roland S-10, works flawlessly. I have tryed on Roland PR-100 also with success.
One idea: durint the write cicle the Roland QD do the cicle Check, Write, Verify. The oled display only displays what he is doing when he is Writing. Perhaps he could display also the other fases (no need for that but would be priety).

By the way I have used the version 2DDFloppy on Roland S-50; S-330 and S-550 with success. Very good.

Congratulations on your work.

AlbDores, 2020-01-07:
And thank you for your priceless support.

keirf, 2020-01-08:
Okay I have updated the wiki thanks!

What fixed your problems? Was it user error, or did the older QD image work better, or what?

AlbDores, 2020-01-08:
QD image work much better.

closed 2020-01-17 by keirf