keirf/flashfloppy #300: Question

omikron88, 2019-12-20:
Hello,

I have the question:

Is Gotek able to send two single density FM data streams at bitrate of 10 us to two output pins? And receive such signals from two input pins?

I am planning to use it as the emulator of Sinclair Microdrive. With proper conversion cable of course.

Microdrive is tape system with an endless loop cartridge. Its runtime is 7s and it has two parallel tracks:
              (bits)
Upper: 7-5-3-1
Lower: -6-4-2-0

Tomas

keirf, 2019-12-20:
You need to check if a timer channel is available on another output pin than RDATA. And another input pin than WDATA. If so you should be fine.

Bazza-666, 2019-12-20:
This would be great 😊

LittleGem, 2019-12-20:
The Gotek has more than enough resources to do this. A full Microdrive emulator can be done with just an ATMega328P :-) check it out on https://vdrivezx.com/vdrivezx. The next iteration of the hardware is using an STM32 but definitely won't be using the Gotek as a hardware platform.

keirf, 2019-12-21:
Yes I'm sure you could do Microdrive on a Blue Pill with SD card attached via SPI. A few dollars in hardware.

Anyway this is out of scope for FlashFloppy, I'm quite unlikely to be adding Microdrive myself.

closed 2019-12-21 by keirf