keirf/flashfloppy #301: FF cannot save to 1.44MB image

bht67, 2019-12-24:
I am unable to get FF to save to 1.44MB disk image on USB stick on Roland DJ-70.  The FF drive can only write to 720KB disk images.  I tried various FF configurations including the one posted in issue #185.  I formatted the blank image using my Roland S-760 and rawread to create a 1.44MB image. 
 How does FF determines whether the image file on the USB is 720KB or 1.44MB?  I need the ability to tell the Roland DJ-70 that the current floppy selected is 1.44MB instead of 720KB.  Any help would be greatly appreciated.  

The Gotek SFR1M44-U100K drive configured as follows:
Jumpers installed on JC and S0
Flashed with FF 3.8a firmware.  

FF configuration set as follows:
host = unidentified
interface = jc
pin2 = dens  (also tried auto w/ same result)
pin34 = chg

keirf, 2019-12-25:
Are you able to read 1.44M images on the Roland? OR do you not have any ready-to-go to try out?

bht67, 2019-12-25:
I was unable to load Roland S-770 images without errors.  It keeps switching between 2 disks at least 10 times, then it asked for a 3rd disk, which I could never find.  Is there a known good 1.44MB image that I could try?

bht67, 2019-12-25:
FF reads and write 720KB perfectly.

bht67, 2019-12-25:
I reconnected the original floppy and verified that the two floppy disks set worked.  It loads much faster and disks switching occurs only once.  I am running out of things to try.  I have other Roland models with floppy drive.  I don't need FF on those because they all support SCSI.  The only two with out SCSI are Roland DJ-70 and Akai X-7000.

keirf, 2019-12-26:
The older ticket you link to does suggest you have the correct FF.CFG. One thing to try is a big downgrade to FF v1.2. This is available still on the FF Download page, and you can downgrade by exactly the same process as firmware upgrade.

I only suggest it because I have had reported on one synth (asr 10) that HD IMG files have not worked since v2.2a, and we haven't tracked down the issue yet!

keirf, 2019-12-26:
Also, Flashfloppy distinguishes between DD and HD images by image file size. The former should be exactly 720k, the latter 1440k. You can also zip and attach useful things like FF.CFG and FFLOG.TXT to this ticket if you like.

bht67, 2019-12-26:
FF v1.2 works.  The DJ-70 can load and save to both 720KB and 1.44MB images.

My DJ-70 works using the following setup:

On Gotek drive:
JC and S0 jumpered

In FF.cfg:
pin2 = dens
Pin34 = chg

On USB stick:
Floppy images had to be renamed with .img file extension.  

Thank you.

keirf, 2019-12-26:
Thanks for the feedback. I think this is the same bug as #290 -- and I now think that density-select on pin 2 (as used by `interface=ibmpc-hdout` and by `pin2=dens`) is not working properly on recent releases for IMG files.

I will sort a fix asap. Would you then be happy to test it for me if I post it here?

bht67, 2019-12-26:
Sure.  I can test out the fixes.  I have a couple of expensive standalone DSO logic analyzers along with Saleae Pro 16 that I recently bought, which I can use if you need me to capture any signals.  

I was thinking the same that pin2 is not signaling correctly.  That is why the DJ-70 only sees the HD image as 720.  I was going to pull out my logic analyzer to verify, but did not get around to it yet.

keirf, 2019-12-27:
Here are fixed versions of v2.x and v3.x. It would be great if you could give them both a quick test:

[ff-301-v2.zip](https://github.com/keirf/FlashFloppy/files/4004635/ff-301-v2.zip)
[ff-301-v3.zip](https://github.com/keirf/FlashFloppy/files/4004636/ff-301-v3.zip)

keirf, 2019-12-27:
This is confirmed fixed.

bht67, 2019-12-27:
Same results as before.  It keeps switching between the 2 disks.

keirf, 2019-12-28:
Why are you using an IMG.CFG? That makes things worse, right :) Get rid of it.

keirf, 2019-12-28:
Also, please now try using the latest 2.xx or 3.xx release (your choice which one) as they both now contain the bug fix.

bht67, 2019-12-28:
The bug fix within 2.x and 3.x are confirmed on Roland DJ-70.  Loading and saving work correctly after I deleted the img.cfg file from the thumb drive.

closed 2019-12-28 by bht67

shindoja, 2023-04-12:
HI again ! Hope you doing well :)
My s760 manage to format one disk but i cant understand what is it.
I just want to be able to make a new disk to put samples on it and i cant since HcX does not let me to.
Each combination i try on the HcX say ' not formatted ' on the s760.
Can you guys tell me what is it then a can make new empty ones ? It says ' Unknown ' in the track analyser

[Drumz.hfe.zip](https://github.com/keirf/flashfloppy/files/11216369/Drumz.hfe.zip)

Thanks a lot !