keirf/flashfloppy #302: USB hub and 2 USB sticks support

Deftaudio, 2019-12-27:
Hi, is there a way to run two USB sticks through USB hub and switch between them? 
I'd like to use one USB as a real one, and the second is from RPI Zero W OTG port for remote file sharing.

keirf, 2019-12-28:
No, not at the moment anyway!

keirf, 2019-12-28:
I'll keep this around for a bit, but it may well end up closed as a wontfix.

StillNotWorking, 2019-12-31:
I can see potential benefits in workflow if this where possible.

Having a USB driver on the RPi or other host computer emulating a memory stick could that be a sensible way to make something like this work? I.e. the host computer only function as a pass through device for the FF to the shared memory stick?

Deftaudio, 2019-12-31:
> I can see potential benefits in workflow if this where possible.
>
> Having a USB driver on the RPi or other host computer emulating a memory stick could that be a sensible way to make something like [...]

That's exactly the idea. Especially considering the cost of Zero W. But I don't want to use it as a single image provider, due to a lag in the startup. Using two FF drives is the only option now.