keirf/flashfloppy #304: Display Order with LCD

bradwh, 2019-12-29:
I have successfully connected a LCD20x04 display to my Gotek running v3.3a release.
This is successfully giving me a display with 
Line 1 Current Folder name.
Line 2 Current image name.
Line 3 Status
Line 4 Image info

If I trying to change the display order from anything but 
display-order = default
will cause the power up initialisation of the Goteck to fail and lockup.
As it works fine when it is set to default, Im guessing I'm not using the display-order syntax correctly.
I would like to set up as follows.
Line 1 Current image name.
Line 2 Image info.
Line 3 Status
Line 4 Current Folder name.

Using the v3.3a firmware can this same display-order setting be used to set up on a LCD20x02
Line 1 Current image name.
Line 2 Image info.

Help on this would be great:=)

keirf, 2019-12-29:
Please update to v3.10a and try again. If you still have problems, please attach your FF.CFG to this issue ticket.

bradwh, 2019-12-29:
All good v3.10a did the trick, now able to configure what is displayed on what line even on a LCD20x2 and also a LCD16x2.
This great for experimenting with different floppy images DS/DD 40T, SS/DD 80T, DS/DD 80T...
my FF.CFG is
display-type = lcd-20x04
display-order = 0,2,1,3
step-volume = 20

closed 2019-12-29 by bradwh