keirf/flashfloppy #305: OLED current dir not refreshed on change to parent

penfold42, 2019-12-30:
Folder name in top row is refreshed on change down, and when i get to / but never on the way up.

Folder structure is this
/games/BloodMoney/BloodMoneyD1of2

My last played game was in / so top row has "/" (good).
select games and top row now has "games" (good)
select BloodMoney and top row now has "BloodMoney" (good).
select [..]
I see the files&folders in /games but top row has "BloodMoney" (NOT! good).
It should display "games"
select [..]
I see the files&folders in / top row has "/" (good).

keirf, 2019-12-30:
Latest 3.10a?

keirf, 2019-12-30:
Yes I can repro this.

penfold42, 2019-12-30:
too funny - i just pulled out the usb stick to double check id updated - yes

keirf, 2019-12-30:
Around src/main.c:1480. Not the clearest code....

keirf, 2019-12-30:
Can you please build and test latest master?

penfold42, 2019-12-30:
sure - will do right now - need a distraction from my GAL nightmares....

penfold42, 2019-12-30:
yep - looks good to me

keirf, 2019-12-30:
Cheers!

closed 2019-12-30 by keirf