keirf/flashfloppy #307: How does one manage SAVE GAMES on Amiga with the Gotek?

muzboz, 2020-01-10:
I recently got a Gotek for my AMIGA.

I've got all the basics set up, but...

I'm unsure of how to manage save games.

I tried putting a blank disk image (made in WinUAE) as Disk "1".

And when I load up King's Bounty, for example, I can't start a new game, because it says there's no disk space on the game disk.

If I try to switch to Disk 1 (blank disk), the game brings up an error saying "Put the game disk back in!"

So, I just can't figure out what I'm meant to be doing to start a game of King's Bounty.

Is there a proven, simple way to manage save games on Amiga with Gotek?

muzboz, 2020-01-10:
I believe I have the same problem with other games, like Gravity Force.

When I finish a Race, set a new time, and it asks for my name, I type in the 3 character high score name, but upon rebooting the game, that's all gone as if it never happened.  The saving to disk doesn't seem to work there either (although no error message is given in that game).

keirf, 2020-01-10:
Are you sure that all these games actually do save high scores to disk? Many arcade-y Amiga games do not!

As for King's Bounty, please try using a blank simple ADF: https://github.com/keirf/FF_Images/blob/master/Amiga/Blank_AmigaDOS.adf

If that doesn't work (some games like Cannon Fodder use a special custom save-disk format) then try this blank HFE image and you may need to format it in-game:
https://github.com/keirf/FF_Images/blob/master/Amiga/Blank_AmigaDOS.hfe

Exactly how game saves work is totally game dependent I'm afraid! But the two blank images above should be all you need when you work out the process for each game,.

muzboz, 2020-01-10:
Thanks very much!  I'll give it a try.  Cheers!

murQla, 2020-01-12:
> I believe I have the same problem with other games, like Gravity Force.
>
> When I finish a Race, set a new time, and it asks for my name, I type in the 3 character high score name, but upon rebooting [...]

Had a problem with Pinball Fantasies saving scores, so i copied game ADF disk image with x-copy to blank HFE image and after that high scores saved flawlessly. Could work lots of other games too.

keirf, 2020-01-12:
Oh that's a good one, when the game saves to the game disk. You see, the game loader gets cracked to amigados, so works in an adf image, but often old cracks wouldnt bother to modify the save game or hiscore routine even if they store in a custom track format. Because that code would still work on a real disk. But it won't work with an adf. I should put this on the faq probably.

muzboz, 2020-01-12:
Thanks for the extra info here.  I was thinking it might be something along these lines, kind of as if the ADF is basically "sort of write protected".  And the way murQla describes things here would explain how to work around the issue.  I'll give it a try some time.

muzboz, 2020-01-12:
I am a little confused about the actual steps to take, though (to ensure that the process works with a Gotek with Flash Floppy).

If at all possible, a working example using a specific game as a test case would be great, describing what programs and steps are needed to create the blank HFE thingy, etc.  

Pinball Fantasies, or King's Bounty would be good ones, I guess, as I think they must both need the game disk to be write-able / save-ready.

I'm only just getting back into the Amiga, so I'm a bit vague on whether you would do the X-Copy process on an emulator, or in WorkBench, or loading X-Copy as an ADF, etc...?  

Can you do it all with just the Gotek, or do you need a DF1 drive as well, or....  etc.  It's just not that simple or clear when you're starting out.  :)

And it seems like something that would help anyone getting started who buys a Gotek and uses Flash Floppy to just play some games, for example.  :D

muzboz, 2020-01-12:
(Apologies if the GitHub is not really the appropriate place for a thread/issue such as this!)

muzboz, 2020-01-13:
REPORTING BACK:  

I put XCopy.adf onto my USB drive.  

I put the Blank_AmigaDOS.hfe you linked me to earlier onto the USB.

I put it into the Gotek, loaded the Selector and assigned those disk images to slots.

Then I loaded XCopy, and used just the Gotek (as Source and Destination) to copy King's Bounty from the original ADF to the Blank HFE.

That all worked fine.

I booted the newly made King's Bounty from the Blank HFE slot.

This all worked fine - King's Bounty loaded.  But when I press "A" to start a new game as the "A" character type, I get the popup box in the game saying, "There is not enough free space on this disk to create a character."

The Blank HFE is 2101 KB.

Not too sure how one is supposed to run this game with the Gotek.

I haven't tried any other games or approaches yet.

muzboz, 2020-01-13:
I tried loading King's Bounty ADF on WinUAE, and I get the same popup message about "not enough disk space" ...  so perhaps I'm just chasing my tail here.  I'll look more broadly into what one is expected to do to play this game at all, anywhere!  :)

muzboz, 2020-01-13:
I just tried King's Quest 1 (original version), and the Save & Load seems to work fine on that, just as an ADF.

So maybe King's Bounty is just a strange game in itself with specific requirements, or the ADF that I happened to have is not a good one.

So maybe I'll go searching for a different ADF of King's Bounty!

Apologies for the run around, but the info above will still come in handy in some cases probably!

muzboz, 2020-01-13:
I just tried a different version of King's Bounty ADF, and it worked fine first go!

So, I may have been leading you on a wild goose chase!  

Thanks very much for the help.

keirf, 2020-01-17:
I have updated the FAQ on the Wiki to address save-game issues.

closed 2020-01-17 by keirf