keirf/flashfloppy #308: Images

Doubletop12, 2020-01-21:
I can't find anything about initialising a USB with ‘n’ preformatted and blank floppy disks ready to be written to. All I can find is a reference to copying images onto the USB but nothing about how these images are created nor how they are copied to the USB.

I must be missing something simple but what is it?

Pete

tenuke, 2020-01-21:
For what machine you need to create the image for?

This is one tool I just googled, so have not tried it myself, dunno if it can do all sort of floppy images or just some specific like for PC or something.
https://www.winimage.com/winimage.htm

Doubletop12, 2020-01-22:
It seems that the initial setup pages are unclear about the use of the HxC tools. They tend to suggest they are only needed if using HxC compatibility mode. If a new user knows nothing of HxC they are unlikely to follow that path and then they have no way of producing an image file on the USB drive.
I think some aspects of the, otherwise excellent, documentation needs reviewing and clarifying for new users.  

I’m using FlashFloppy modified Gotek on a homebrew CP/M system and have struggled a bit to get my head around what is required. I’m not there yet but the system can see the drives and I’m able to format them. I’m getting errors but will work through what the causes of the problems are. 

I was wondering how I could transfer files to and from a PC through the Gotek to the CP/M system but it occurs to me that as the HxC software is unlikely to know the disk layout and isn’t going to be able to help.

keirf, 2020-01-22:
Yes, HxC software is not a panacea. It will do conversions between disk-image formats, and it understands FAT filesystems I believe, but it's not going to help you access CP/M files.

Your best bet on FlashFloppy is probably to use blank HFE image files. I suggest a blank DSDD image like this one: https://github.com/keirf/FF_Images/raw/master/Amiga/Blank_AmigaDOS.hfe

Format it to CP/M format on your CP/M system.

Doubletop12, 2020-01-22:
Keir

Thanks for the suggestion. I think that’s what I'm doing, I used the HxC software to produce a blank image 40 tracks, 9 Sectors/track 512 byte sectors double sided. The trouble with it was that the sectors weren't initialised as "e5" that CP/M expects. But my CP/M format programme fixed that. I think.

I need to tweak some of the CP/M BDOS disk parameters to match as there were some slight differences to the original values I'd set up 30 years ago.

I'd also come to the conclusion that some conversion software was needed on the PC side but one of the many CP/M emulators may be able to do that If I use my BDOS as the basis.

I do believe that your pages would benefit users if there were some clues about preparing the USB. It’s a bit light and I had to search around to discover a way forward and even then I’m not convinced I’m doing the right thing

Pete

Doubletop12, 2020-01-23:
There may be other options but this is working for me for CP/M disks

[http://www.classiccmp.org/cpmarchive...ools/index.htm](url)

Pete

closed 2020-01-30 by keirf