keirf/flashfloppy #311: Tatung Einstein 256 dual floppy support

simulant-systems, 2020-01-28:
Hi I recently got a Tatung Einstein 256 which has the 3" floppy drive (like Amstrad computers used and some others).

There is very little info on the machine that I can find online, but a few magazines and books I have mention the computer supports an external 2nd floppy drive. As my internal drive is good, I'd love to add a Gotek as the external.

My problem is there is just one connector for a floppy on the pcb!! And the internal floppy uses this. My reference material states that to add a second drive you must "unplug the intermal floppy drive from the floppy connector, plug the external drive ribbon into the connector, and then plug the internal drive into the adapter on the external ribbon".

I can't find any docs or other info on what this "pcb adapter" may be or look like...

Do you have any idea how I may be able to make an adapter or figure out the pinout? I do have the full schematics of the computer in its manual if they may help establish how to connect two drives to one plug??

Should I just make a ribbon with two connectors on and try it like a modern PC works with everything on the two drives going to the same pins but set master and slave (sl1) on the floppy and gotek? This sounds too simple as the minimal docs I have on external drives mention the adapter pcb...

Thanks for any help at all, it is a dream project for me to add an external drive to the Einstein 256. Thanks again.

penfold42, 2020-01-28:
Do you have the schematics for the computer ?

Apparently it uses a WD1770 controller chip so hopefully most of the signals are sane and shugart like.
If you’re lucky the adaptor board will just shuffle or gate a couple of drive select lines.

weirdocollector, 2020-01-29:
I too have a Tatung 256 in my collection and I have got the very same doubts about attaching it a second drive.

The only element I can add is that a Gotek with Flash Flopy works perfectly on my Tatung Einstein, the 256 predecessor, which uses the same 3" drives.

I'll follow this thread with genuine interest.

simulant-systems, 2020-02-22:
Hi, I finally dug out the manual so I could scan in and post the schematic, I hope someone can make sense of how to hook up the external floppy from this...

[image]

penfold42, 2020-02-23:
it looks like a standard collection of pins but a pin arrangement I’m not familiar with (but I’m not familiar with many!)

The normal pins go to the wd1770 floppy controller and there are two drive select lines (DS0 and DS1) that go to a different chip.

Do the drives themselves have drive select jumpers ?

simulant-systems, 2020-05-09:
Sorry for the slow reply! Well the computer seems to use the standard 3" drive that Amstrad PCW and Amstrad CPC6128 computers also use. I own Amstrads too and can't see any differences with the drives - none of which have Drive Select Jumpers physically on the drive itself...

Here is a list of all the computers that use this model of 3" disk drive:

- AMDECK (AMDISK) [1982] : external drive for TRS-80 , IBM PC , Atari 400/800 , Apple II
- YAMAHA MDR-1 [1983] : sequencer for professional organs. Pre-MIDI 
- National MyBrain 3000 [1983] : computer
- Gavilan Laptop [1983]
- Tatung Einstein [1984]
- Oric Atmos external Disc Drive [1984]
- Sega SF-7000 Disc Drive (for Sega SC-3000 computer) [1984]
- Timex FDD / FDD-3000 drive ( for ZX Spectrum and Timex ) [1985]
- Amstrad / Schneider CPC [1985]
- Amstrad / Schneider PCW (8000 series) [1985]
- Sinclair Spectrum +3 [1987]
- Amstrad PCW 9000 series [1987]
- Amstrad CPC+ [1990]

I wonder if any of these other computers also just have one connector for a floppy drive. I know Amstrads have ports on the back for external drives...

I'll ask my friend from Bryce from Amstrad circles (he is the MAN when it comes to Amstrad hardware) to see if he has any ideas on this that he may be able to post here. Thanks for the help so far :)

closed 2020-05-29 by keirf

simulant-systems, 2020-05-31:
Ok sorry to reopen but I confirm that FlashFloppy works great as both first and second drives on Tatung Einstein 256 (internally and externally) and TC-01 too. Bryce got back to me and confirmed that 3" Amstrad drives don't need jumpers or extra motherboard ports, they can just just work as master/slave by just using a double ribbon with two connectors on them. I've tested this on the Tatungs and it works great. You're fine to reclose this ticket I just wanted to tell you that Tatung Einstein computers work well :)