keirf/flashfloppy #312: FF V8-FF hang at title screen.

nicr4wks, 2020-01-30:
Hi All,

I have this connected to an Amiga 500.

I've just updated my gotek for the first time in 2 years and have hit an issue with the FF hxc selector hanging at the main screen that shows the facebook/email links. The FW Field in the bottom left constantly shows ----- and the segment displays d-A.

Pressing mouse, gamepad, enter, esc, help, del makes no difference.

I followed this process:
Updated to FF_Gotek-Bootloader-v2.14.upd
Then updated to FF_Gotek-v2.14.upd

Copied HxC_Compat_Mode\HXCSDFE.CFG and Amiga\AUTOBOOT.HFE to the USB drive.
Plugged USB drive in to gotek and rebooted, the loader shows the main title screen OK but hangs as described above.

The following combination works perfectly fine:
FW FF-V2.14
HxC Selector v1.89a

Thanks!

keirf, 2020-01-30:
There are earlier releases of the FF fork for download here: https://github.com/keirf/HxC_FF_File_Selector/releases

Could you try one or two of them to find out if it is a recent problem? Where the download link is not obvious, it's hidden under an "Assets" tab.

keirf, 2020-01-30:
Just tested via your process (updated to 2.14 bootloader, 2.14 main firmware, then run v8-FF selector)... Works for me, on my A500 test board.

Does your OP imply that actually any version of FF-HxC selector is at fault (you tried an old one first, then a new one)?

nicr4wks, 2020-01-31:
Just had time to do some more in depth testing.

v7-FF seems to be working OK.
v8-FF seems to be intermittent, I hard reset the a500 from the power brick 10 times, 3 times it loaded OK the other 7 times it hung at the title screen. I've tested with 2 freshly formatted USB sticks and also ran the TestKit image over the ram in case something was bad there (no errors)

keirf, 2020-01-31:
The only fixes in v8-FF are for Atari ST. I can *guarantee* (just checked) that the only differences in the Amiga executable between v7-FF and v8-FF are the version-number and date-time strings. It is therefore an intermittent issue in hardware or software: I'd be surprised if you couldn't reproduce it on v7-FF if you try for long enough.

It's difficult to tell where the hang is happening without doing some debug builds to print stuff on screen or via serial port. Or using an Action Replay cartridge to break into execution and look at the machine state.

keirf, 2020-02-05:
Without more info or corroboration I don't think there's much I can do here.

closed 2020-02-05 by keirf