keirf/flashfloppy #313: Flashfloppy 2.14 not working with Sam Coupe

o0Nosferatu0o, 2020-02-02:
Hi Keirf, my name's Pedro Torres and i'm from Portugal. I was attempting to use a Gotek with my Sam Coupe. The Sam has 3.0 Rom, 512k ram and an original floppy drive.
This Gotek is working without a problem with Amiga/Amstrad CPC/Atari. It was flashed to 2.14 and couldn't boot any .dsk with it. I came upon the https://github.com/keirf/FlashFloppy/issues/92 and tested the #2 firmware you posted there and it worked right of the bat. With 2.14 started to read and stated "53 No Dos" stoping at track 04.

keirf, 2020-02-02:
Could you send me a link to, or attach here, a DSK which works now but did not work in 2.14.

Also could you try the following firmwares available on the Downloads page:
- v0.9.18a (This I guess is the proper official release on which the one you downloaded was based)
- v0.15
- v1.2

That will do for starters :) Thanks!

keirf, 2020-02-25:
Reopen if anyone can reproduce this issue and supply the requested info.

closed 2020-02-25 by keirf