keirf/flashfloppy #316: Problem with the rotary control

briefer666, 2020-02-15:
Hi

I have installed a KY-040 rotary control, and it works but only to move forward through the menus. Moving the rotary clockwise or anticlockwise have the same effect, move to the next item, not the previous.

I try to remove the pull-up resistors from the back of the PCB and with and without plugin the +3.3v pin, but did not fix the issue.

any clues? should I try soldering directly to the rotary directly?

kind regards

penfold42, 2020-02-15:
Did you also hook up the push button ?

Is it possible you switched a couple of wires ?

keirf, 2020-02-15:
Flashfloppy version? Have you tried adjusting rotary= setting in FF.CFG?

briefer666, 2020-02-15:
Thank you for your quick answer. Have to look in my browser history, but it seems like I installed 3.7a (I installed it some time ago and I am trying to install the rotary now)

I checked the wires two and three times, everything seems to be in place.

[url=https://ibb.co/cJz5QTd][img]https://i.ibb.co/5TqDLcV/IMG-0084.jpg[/img][/url]
[url=https://ibb.co/dmsxShm][img]https://i.ibb.co/xFKdy4F/IMG-20200213-140930.jpg[/img][/url]

I also tested with FF.CFG with rotary=reverse,half, and then it only goes back, not forward.

Should I try another fw?

keirf, 2020-02-15:
Try latest 3.x and if no luck with any rotary= setting I think you need a new rotary encoder.

briefer666, 2020-02-15:
Thanks! I will do that 😉

briefer666, 2020-02-15:
Thanks! I will do that

briefer666, 2020-02-15:
Ok, I updated and it went to worse :( 

now the rotary does not move through the menus. I tried rotaryas full, quarter, half and grey. Only quarter and grey moves, but goes to the next item and back again in one step.
Push button always work.

I am going to solder the cables directly to the rotary contacts, and bypass the board.

cheers

briefer666, 2020-02-15:
And it works! wiring directly to the rotary bypassing the board fixed the problem. Thanks!

closed 2020-02-15 by briefer666